lgsvl/simulator #1686: Installation of PythonAPI not working

marcusvinicius178, 2021-08-15:
Hi I am following the tutorial: https://www.svlsimulator.com/docs/python-api/python-api/

However it is not working. The system is unable to localize the module environs:

[image]

On my machine I have installed months ago and it is ok...but in the rented cloud machine that I use, which is more powerful, it is not working...

@EricBoiseLGSVL  dou you have some idea how can I fix this? I have already tried use a virtual environment but it is not working...

heeen, 2021-08-16:
you are using sudo to invoke pip3, Try using pip3 without sudo.

roshambo919, 2021-10-06:
@marcusvinicius178 do pip install using the instructions here
https://github.com/lgsvl/PythonAPI

from requirements.txt and not setup.py.

```
python3 -m pip install -r requirements.txt --user .

# install in development mode
python3 -m pip install -r requirements.txt --user -e .
```

closed 2021-10-06 by EricBoiseLGSVL