lgsvl/simulator #1688: Exception while running a 14-create-pedestrians.py Quickstart Script (PythonAPI)

roboserg, 2021-08-16:
Latest version of everything, tried some quick-start scripts, they worked, several didn't. One of them is `14-create-pedestrians.py`, throwing an exception `Exception: Object reference not set to an instance of an object`

`Python API Quickstart #14: Creating pedestrians on the map
Press Enter to create 100 pedestrians
(1) adding Stephen
Traceback (most recent call last):
  File "e:\DeepLearning\SVL Sim\01_Hello\PythonAPI\quickstart\14-create-pedestrians.py", line 57, in <module>
    p = sim.add_agent(name, lgsvl.AgentType.PEDESTRIAN, state)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\simulator.py", line 162, in add_agent
    uid = self.remote.command("simulator/add_agent", args)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object`

roboserg, 2021-08-16:
**2.** Another issue I have is that some quick-start scripts load the CubeTown1 map and every script doing this doesn't work me, the simulation keeps stuck after the first print message in these scripts. Concrete example, right now I have the same CubeTown1 loaded (it's empty). I run` 01-connecting-to-simulator.py` and run the following simple code.

The only thing that is printed is `Python API Quickstart #1: Connecting to the Simulator`, it seems stuck on getting the env or simulator instance. The `env` variable is empty. It doesn't seem to connect. How can I restart the environment programmatically, without having to manually stopping the simulation in SVLSimulator and why does it happen?

```
from environs import Env
import lgsvl

print("Python API Quickstart #1: Connecting to the Simulator")
env = Env()

sim = lgsvl.Simulator(env.str("LGSVL__SIMULATOR_HOST", lgsvl.wise.SimulatorSettings.simulator_host), env.int("LGSVL__SIMULATOR_PORT", lgsvl.wise.SimulatorSettings.simulator_port))

sim.run(1)
print(sim.status)
print("Version =", sim.version)
print("Current Time =", sim.current_time)
print("Current Frame =", sim.current_frame)
```

roboserg, 2021-08-16:
**3.** There seems to be something wrong going on with the Python API on my end. Similar to issue #2 from above, [my simple python script](https://hatebin.com/gckikhjlpc) runs fine once. If I restart the script again, similar to issue #2, the map loads and nothing happens afterwards. I guess the Sim breaks. I tried using both the compiled SVL sim 2021.2.1 from the release tab AND I also used local Unity 2020.3.16f install and run from the editor. 

If using the editor with Unity 2020.3.16f I get an exception in python: `Exception: Object reference not set to an instance of an object`

```
Starting env...
unknown
Spawning agent
Spawning NPC

Traceback (most recent call last):
  File "e:\DeepLearning\SVL Sim\01_Hello\PythonAPI\quickstart\test.py", line 22, in <module>
    npc = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\simulator.py", line 162, in add_agent       
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object
```

roboserg, 2021-08-17:
Tried with Unity 2020.3.3f, same issue, the sim dies when running [my script](https://hatebin.com/gckikhjlpc) for a second time. This is reproducible endlessly. 
[image]

EricBoiseLGSVL, 2021-08-18:
Hello.  Glad you like svl simulator.  First off, let's create separate issues for each issue instead of one for many.  This helps other users to find answers if they have the same problem.

1) Did you build pedestrians and npcs and have them in the AssetBundle folder?  We have moved these assets to separate repos to reduce dependencies in core.  You can find then in the same release tag binaries we release and you can just copy over to the simulator repo AssetBundles folder

2) I'm not sure what the issue is here.  This quick start script doesn't load a map.  It should just have console output.
[image]
 
3) From a quick glance at your .py class, you don't reset on env load.  Please see all quick scripts we have provided and how we do this.  It is required to reset simulations.

roboserg, 2021-08-19:
@EricBoiseLGSVL 
Thanks Eric for the info.

1. Is resolved, no clue what the issue was, the script works now

2 / 3. I [modified my code ](https://hatebin.com/buqqrmhrcd) to actually reset the scene. It works better now. So is loading calling "load scene" if the same scene is already loaded, without using `scene.reset()` will always cause a crash and this is by design? I mean, I would think the simulator should not care if it has the current scene already loaded, it seems weird it crashes when one loads the same scene again. Video of simulator crashing when loading the same scene again:

https://user-images.githubusercontent.com/4758917/129993498-c8202450-bcfe-4a9d-b20a-c53cc76887a7.mp4

4. When the simulator is crashed as above, is there any programmatic way from the Python API to "reset" or "fix" the simulator? The *.exe binary is running and working, but it doesnt respond to any Python commands.

EricBoiseLGSVL, 2021-08-19:
1-3) Glad this is resolved.  Let us know if it happens again and you can debug the cause.
4) Reset is required because we collect analysis data to display for you in webui and to run in cloud simulations.  We also need to manually reset to make sure all coroutines and logic is cleaned up before starting a new sim.  It also ties into AD stacks logic that needs to be fully reset to run another simulation.
I'm not sure why you would want a fix, the exceptions are there on purpose to stop simulation execution. This is done by design. If you want a reset, you just need to call it. What would be the use case? I feel I don't quite understand what you are asking for.

roboserg, 2021-08-19:
4) Thanks for the elaborate answer. It will probably save some hours for the next beginners if `scene.load()` would perform a `scene.reset()` automatically. So maybe moving the if checks that we do in the scripts in the body of `scene.load()`?

EricBoiseLGSVL, 2021-08-19:
Good point, we will look into this and see what this change will effect.  Thanks for posting, this helps.

closed 2021-08-19 by roboserg