lgsvl/simulator #1689: Issue with git cloning the 2021.1 version

roboserg, 2021-08-16:
Unity 2021.2 seems in beta now, so the last stable Unity is 2021.1.17f. I wanted to git clone the corresponding VSL sim branch `release-2021.1` with `git clone --single-branch --branch release-2021.1 https://github.com/lgsvl/simulator.git` but it downloads the 2020.3 version, not 2021.1. I tried the same for 2021.2 of SVL sim branch, same story - Unity 2020.3. Git clone master - 2020.3 again.

How do I get the 2021.2 or 2021.1 Unity versions of SVL sim?

roboserg, 2021-08-16:
Git branch shows as being correct, but Unity Hub says its the old Untiy version https://i.imgur.com/gIJSWsB.png

lemketron, 2021-08-16:
The SVL version number is not the same as the unity version number.  

Please refer to the documentation (or readme) to know which version of Unity is required for a given branch/version of SVL. 

SVL Simulator 2021.2 requires **Unity version 2020.3.3f1** as noted in the ["Building from source" documentation](https://www.svlsimulator.com/docs/installation-guide/build-instructions/).

roboserg, 2021-08-16:
Thanks! Yea, I read the documentation about "building from source", I thought the documentation is outdated.

> Please refer to the documentation (or readme) to know which version of Unity is required for a given branch/version of SVL.

I can't find any Unity version number mentioned in the README and the documentation is moved off GitHub, so it's unclear what version of Unity to install for which branch. Readme would be definitely be the best place for it due to convenience and automatic source control.

lemketron, 2021-08-17:
You can find the documentation for older versions in the [documentation archive](https://www.svlsimulator.com/docs/archive/) including build instructions. 

The archived [2020.2 build instructions](https://www.svlsimulator.com/docs/archive/2021.2/installation-guide/build-instructions/) specify Unity version 2020.3.3f1.
The archived [2020.1 build instructions](https://www.svlsimulator.com/docs/archive/2021.1/installation-guide/build-instructions/) specify Unity version 2019.4.18f1.
The archived [2020.1.1 build instructions](https://www.svlsimulator.com/docs/archive/2021.1.1/installation-guide/build-instructions/) specify Unity version 2019.4.18f1.

You can download the Unity versions on the [Unity download archive](https://unity3d.com/get-unity/download/archive).

closed 2021-08-17 by roboserg