lgsvl/simulator #1690: How to control the simulator car using autoware_msgs/VehicleCmd

lorlor, 2021-08-18:
Hi guys,
When I tried to control simulator cars using autoware_msgs/VehicleCmd through rosbridge, I got these errors like this.
[image]
I do not figure out where I did wrong.

And my environment is:
OS: Ubuntu 16.04
Autoware Version: 1.12
Simulator version: 2021.1.1

Any advise will be thankful.

EricBoiseLGSVL, 2021-08-18:
Can you please post the player.log to look at all errors.  Most likely it is an issue with a connecting to the bridge.  Do you have it running before you run the simulation?

lorlor, 2021-08-19:
Thanks for your reply, @EricBoiseLGSVL . I am pretty sure that rosbridge has been launched before running the simulator. 
Here are the log info. It seems like there was something wrong about websocketsharp.

8/19/2021 3:07:36 PM|Error|WebSocket.messagec:0|System.NullReferenceException: Object reference not set to an instance of an object
                             at Simulator.Sensors.AutowareVehicleControlPlugin.ConvertMsg (Simulator.Bridge.Ros.Autoware.VehicleCmd data) [0x00004] in <5d37dcaa1252412c96d89bb9f7e39bac>:0 
                             at Simulator.Bridge.Ros.RosBridgeFactoryBase+<>c__DisplayClass3_1`2[DataType,BridgeType].<RegSubscriber>b__1 (SimpleJSON.JSONNode rawData) [0x00017] in <0dd23c470794411eaff2d96778f91e6d>:0 
                             at Simulator.Bridge.Ros.RosBridgeInstance.OnMessage (System.Object sender, WebSocketSharp.MessageEventArgs args) [0x00091] in <0dd23c470794411eaff2d96778f91e6d>:0 
                             at WebSocketSharp.Ext.Emit[TEventArgs] (System.EventHandler`1[TEventArgs] eventHandler, System.Object sender, TEventArgs e) [0x0000b] in <84d0020558774b6c88b3203b756a09ec>:0 
                             at WebSocketSharp.WebSocket.messagec (WebSocketSharp.MessageEventArgs e) [0x00003] in <84d0020558774b6c88b3203b756a09ec>:0  
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

An error has occurred during an OnMessage event. 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

NullReferenceException: Object reference not set to an instance of an object
  at Simulator.Sensors.AutowareVehicleControlPlugin.ConvertMsg (Simulator.Bridge.Ros.Autoware.VehicleCmd data) [0x00004] in <5d37dcaa1252412c96d89bb9f7e39bac>:0 
  at Simulator.Bridge.Ros.RosBridgeFactoryBase+<>c__DisplayClass3_1`2[DataType,BridgeType].<RegSubscriber>b__1 (SimpleJSON.JSONNode rawData) [0x00017] in <0dd23c470794411eaff2d96778f91e6d>:0 
  at Simulator.Bridge.Ros.RosBridgeInstance.OnMessage (System.Object sender, WebSocketSharp.MessageEventArgs args) [0x00091] in <0dd23c470794411eaff2d96778f91e6d>:0 
  at WebSocketSharp.Ext.Emit[TEventArgs] (System.EventHandler`1[TEventArgs] eventHandler, System.Object sender, TEventArgs e) [0x0000b] in <84d0020558774b6c88b3203b756a09ec>:0 
  at WebSocketSharp.WebSocket.messagec (WebSocketSharp.MessageEventArgs e) [0x00003] in <84d0020558774b6c88b3203b756a09ec>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Bridge.Ros.RosBridgeInstance:OnError(Object, ErrorEventArgs)
WebSocketSharp.Ext:Emit(EventHandler`1, Object, ErrorEventArgs)
WebSocketSharp.WebSocket:error(String, Exception)
WebSocketSharp.WebSocket:messagec(MessageEventArgs)
WebSocketSharp.WebSocket:message()
WebSocketSharp.<>c__DisplayClass174_0:<startReceiving>b__1(WebSocketFrame)
WebSocketSharp.<>c__DisplayClass84_0:<ReadFrameAsync>b__3(WebSocketFrame)
WebSocketSharp.<>c__DisplayClass77_0:<readPayloadDataAsync>b__0(Byte[])
WebSocketSharp.<>c__DisplayClass59_0:<ReadBytesAsync>b__0(IAsyncResult)
System.Net.Sockets.<>c:<Complete>b__27_0(Object)
System.Threading.QueueUserWorkItemCallback:System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
System.Threading.ThreadPoolWorkQueue:Dispatch()
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()
 
(Filename: <5d37dcaa1252412c96d89bb9f7e39bac> Line: 0)

And in a terminal, I could get the nmea_sentence information from simulator

lemketron, 2021-08-19:
What is the Topic in your AutowareAi Control Sensor configuration?  It should be /vehicle_cmd but our [documentation](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#autowareai-control) for this sensor has incorrect topic information in the list of JSON parameters.

EricBoiseLGSVL, 2021-08-19:
If it is websocket sharp and not what @lemketron posted, we have a update in the 2021.3 release.  I suggest trying that when we push to master soon to test if it fixes this issue.

cyberphysicalrobotics, 2021-08-19:
I tested

lorlor, 2021-08-23:
Thank you, @lemketron . The topic name is configured as /vehicle_cmd. I will test the AutowareAi 1.14.

lorlor, 2021-09-01:
I tried the Autoware1.14, but I still got the same error information. Should I change the linux version or ros version to make more test?

revati-naik, 2021-09-03:
@lorlor You do not need to change the linux version or ROS version in that case. You can use the Autoware.AI docker container. Please follow the steps as show [here](https://www.svlsimulator.com/docs/archive/2020.06/autoware-instructions/). Clone the docker repository from autoware.ai into a working directory. 

Then run the docker container as mentioned in the steps. It will use the Autoware.Ai 1.14 version.

lorlor, 2021-09-07:
Thank you @revati-naik. But I need to use Autoware.AI without docker. I have a question about this. Could the autoware_msgs/VehicleCmd data be parsed in simulator. Because I doubt that the websocketsharp module of simulator maybe not parse the autoware_msgs/VehicleCmd data.

And this is my test 'rostopic pub':

$ rostopic pub /vehicle_cmd autoware_msgs/VehicleCmd '{header: auto, steer_cmd: {header: auto,steer: 0}, accel_cmd: {header: auto, accel: 0}, brake_cmd: {header: auto, brake: 0}, lamp_cmd: {header: auto, l: 1, r: 1}, gear: 1, mode: 1, twist_cmd: {header: auto, twist:{linear: {x: 10 , y: 10, z: 10}, angular: {x: 5, y: 5, z: 5}}}, ctrl_cmd: {linear_velocity: 3, linear_acceleration: 3, steering_angle: 0.1}, emergency: 0}'

Before sending this test command, I have confirmed that I could get simulation data like GPS sentences using rosbridge_websocket

lorlor, 2021-09-07:
My ros distribution is kinetic. Will this fail the command sending?@EricBoiseLGSVL @cyberphysicalrobotics @revati-naik