lgsvl/simulator #1692: Other way of downloading Map and Vehicle in Simulator

DeepCollision, 2021-08-18:
Hi team, 

Is there any other way to download the map and vehicle for simulator, due to the network speed is too slow on my side, my download is very slow  when using the follow way.
[image]

EricBoiseLGSVL, 2021-08-18:
No, sorry.  We are looking to expand coverage of the servers.  Where are you located?

mark-gerow-lge, 2021-08-18:
Technically there are two workarounds you can do if sufficiently motivated

1) Clone the project (not the binary) and any maps/vehicles you want that we provide the source to, and turn on Developer Debug Mode in Simulator->Developer Debug Mode in Unity Editor. This will load assets you have locally in Assets/External/{type] rather than downloading if their asset name matches the name of what you set in WISE.

2) Manually add the assets to the appropriate location and rewrite the data. You should be able to find both at

Windows
%userprofile%\AppData\LocalLow\LGElectronics\SVLSimulator

Linux
$XDG_CONFIG_HOME/unity3d or $HOME/.config/unity3d

Environments go in the Environments subfolder, Vehicles go in the Vehicles folder, and you have to modify the Assets table to contain your manually added assets. You can find the asset guid by opening up your asset bundle in a zip reader and reading the manifest. Obviously not ideal but if you cannot tolerate downloads from WISE due to connection you can save a lot of download time between these two methods.