lgsvl/simulator #1693: How to prevent the passing maneuver to happen in the sim?

krtt, 2021-08-18:
I'm using the Python API sim and py scripts to create test scenarios. I need to create a test scenario where the ego vehicle stops for a pedestrian who is on a crosswalk. The ego should not pass the pedestrian but stop before it and stay in the first lane. The problem is that the ego passes the pedestrian and keeps driving. 

Question 1: How to prevent the ego vehicle to pass NPC vehicles and pedestrians in the sim, using python?

Question 2: How to prevent the ego vehicle leaving the first lane in some scenarios, using python?

Video clip showing the current passing maneuver:

https://user-images.githubusercontent.com/22684441/129895677-d20b9e2e-b127-4b22-9a44-16c5bf9077cb.mp4

Ubuntu 18.04
Sim v2021.2
Apollo 6.0

EricBoiseLGSVL, 2021-08-18:
What are you using for logic for the ego vehicle to pass?  Looks like you need to check obstacle type.  If it is an NPC, pass.  If it is a pedestrian, stop and wait.

lemketron, 2021-08-19:
If Apollo is controlling the vehicle then it is up to (e.g. the planner module in) Apollo to decide what the ego should do when it encounters a pedestrian or another vehicle. 

You cannot script or otherwise control the ego behavior when it is controlled by an external autopilot (like Apollo).

EricBoiseLGSVL, 2021-08-19:
Thanks @lemketron , yeah I was confused because it was great passing logic :) If you wrote an algorithm in python to do this, please share with the community.

krtt, 2021-09-25:
The solution is in Apollo indeed. The parameter is called `allow_lane_borrowing`. In the apollo/modules/planning/conf/scenario/ directory there are many files with this parameter and it seems to be by default true. You can change it to false where needed.

For example, for the simple lane following functionality make the change in the file **lane_follow_config.pb.txt** and set it to false:
`allow_lane_borrowing: false`

That's all. The ego will not swerve anymore when doing lane following in the sim.