lgsvl/simulator #1694: why random traffic not working with my lanelet2 HD map?

ljjhome, 2021-08-18:
Hi, I'm new to lgsvl. I've been working on the HD map part for two days.
I use JOSM to build the lanelet2 map suggested by [this](https://github.com/fzi-forschungszentrum-informatik/Lanelet2).
I have read the #1582 . and I'm sure the NPC asset bundles are fine.

At first, I built a big map, and it works well with lanelet2 API. But when I import it to LGSVL, no random traffic appears. 

Then I built a small map, just like the built-in map "Straight2LaneOpposing", I used the same map annotations. and it works well with random traffic. 

Then I modify the map as shown below ( add a small lanelet at upper right corner) : 
[image]
It works well with random traffic. Both lanes have NPS vehicles

but when I add another lanelet at upper right corner:
[image]

Then there is no random traffic on the lane to which the new small lanelet is added. but the random traffic works well on the other lane. 

I'm wondering: is it because I draw the map in the wrong way 
                       or is there something else or other settings I didn't notice. 

system : Ubuntu 18.04
software : 2021.2.1
Thank you

EricBoiseLGSVL, 2021-08-18:
Can you please show the annotation in the Unity scene not the lanelet2 viewer?

ljjhome, 2021-08-19:
Thanks for the reply, I draw two straight lanes this time. as below: I follow the steps in [this](https://www.svlsimulator.com/docs/simulation-content/hd-map-mesh-generation/)
[image]

when I import it to lgsvl, the result is like below:
[image]

there is random traffic behind the ego vehicle, but there is no random traffic on the other lane.

This is the lanelet2 file I download after upload to the web:

﻿`<?xml version="1.0" encoding="UTF-8"?><osm version="0.6" generator="LGSVL Simulator"><node id="1" lat="22.6826531696899" lon="114.196427838487" visible="true" version="1"><tag k="x" v="6.000652" /><tag k="y" v="3.56337" /><tag k="ele" v="0" /></node><node id="2" lat="22.6828895868645" lon="114.196843720666" visible="true" version="1"><tag k="x" v="31.38573" /><tag k="y" v="-39.68641" /><tag k="ele" v="0" /></node><node id="3" lat="22.6831099171827" lon="114.197235463261" visible="true" version="1"><tag k="x" v="55.03552" /><tag k="y" v="-80.42071" /><tag k="ele" v="0" /></node><way id="4" visible="true" version="1"><nd ref="1" /><nd ref="2" /><nd ref="3" /><tag k="type" v="line_thin" /><tag k="subtype" v="solid_solid" /><tag k="color" v="yellow" /></way><node id="5" lat="22.6825996333312" lon="114.196463589054" visible="true" version="1"><tag k="x" v="0" /><tag k="y" v="0" /><tag k="ele" v="0" /></node><node id="6" lat="22.6828359722037" lon="114.196879333302" visible="true" version="1"><tag k="x" v="25.37668" /><tag k="y" v="-43.23545" /><tag k="ele" v="0" /></node><node id="7" lat="22.6830562248049" lon="114.197270937586" visible="true" version="1"><tag k="x" v="49.01814" /><tag k="y" v="-83.95537" /><tag k="ele" v="0" /></node><way id="8" visible="true" version="1"><nd ref="5" /><nd ref="6" /><nd ref="7" /><tag k="type" v="line_thin" /><tag k="subtype" v="solid" /><tag k="color" v="white" /></way><relation id="9" visible="true" version="1"><member type="way" ref="4" role="left" /><member type="way" ref="8" role="right" /><tag k="location" v="urban" /><tag k="subtype" v="road" /><tag k="participant:vehicle" v="yes" /><tag k="type" v="lanelet" /></relation><node id="10" lat="22.6827067060375" lon="114.196392087891" visible="true" version="1"><tag k="x" v="12.0013" /><tag k="y" v="7.12674" /><tag k="ele" v="0" /></node><node id="11" lat="22.6829432015493" lon="114.196808108038" visible="true" version="1"><tag k="x" v="37.39478" /><tag k="y" v="-36.13738" /><tag k="ele" v="0" /></node><node id="12" lat="22.6831636095853" lon="114.197199988981" visible="true" version="1"><tag k="x" v="61.05291" /><tag k="y" v="-76.88605" /><tag k="ele" v="0" /></node><way id="13" visible="true" version="1"><nd ref="10" /><nd ref="11" /><nd ref="12" /><tag k="type" v="line_thin" /><tag k="subtype" v="solid" /><tag k="color" v="white" /></way><relation id="14" visible="true" version="1"><member type="way" ref="4" role="left" /><member type="way" ref="13" role="right" /><tag k="location" v="urban" /><tag k="subtype" v="road" /><tag k="participant:vehicle" v="yes" /><tag k="type" v="lanelet" /></relation></osm>
`

EricBoiseLGSVL, 2021-08-19:
This may be a spawning issue.  Can you share the MapOrigin settings you have?  You can try turning on some options there like so.
[image]

ljjhome, 2021-08-19:
Thank you for the advice, I tried turning on these options and they do help.

this is my MapOrigin setting, and it works now. 
[image]

could you please show me some document on how exactly these options affect the random traffic?

 I'm going to try drawing some big maps, can we just leave the issue open unitl I succeed with big maps

EricBoiseLGSVL, 2021-08-19:
These options effect spawning.  We have them for small maps or maps that don't have edges.
- Ignore NPC Visible = remove all frustum checks for spawns (e.g. ignores that the camera can see the npc on spawn)
- Ignore NPC Spawnable = remove checks for only spawning on MapLanes that don't connect (e.g. CubeTown)
- Ignore NPC Bounds = remove checks for NPC Spawn Bounds Size (e.g. ignores spawn bounds on ego)

ljjhome, 2021-08-20:
Thank you for the detailed explanations. They are really helpful.