lgsvl/simulator #1695: Exception when using add random agent for the pedestrian

roboserg, 2021-08-19:
[Full source code](https://hatebin.com/epdnaubjzl)

In the code, Line 17 `npc = sim.add_random_agents(lgsvl.AgentType.PEDESTRIAN)` crashes with:

```
Spawning PEDESTRIAN
Traceback (most recent call last):
  File "e:\DeepLearning\SVL Sim\01_Hello\PythonAPI\quickstart\test.py", line 26, in <module>
    npc = sim.add_random_agents(lgsvl.AgentType.PEDESTRIAN)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\simulator.py", line 178, in add_random_agents
    self.remote.command("simulator/add_random_agents", args)
  File "e:\deeplearning\svl sim\01_hello\pythonapi\lgsvl\remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object
```

Simulation breaks too. Maybe pedestrians need a sidewalk to be spawned?

EricBoiseLGSVL, 2021-08-23:
Yes, the map needs nav mesh and pedestrian annotation.  We have a fix for next release with maps that have nav mesh but no pedestrian annotation.
Do you have pedestrians in the AssetBundles folder?

roboserg, 2021-08-23:
> Do you have pedestrians in the AssetBundles folder?

Yes, I do (using the compiled binary from the latest release). They do work in your example scripts, but not with my script. 
How do I know the map doesnt have pedestrian annotations? No side walks?

EricBoiseLGSVL, 2021-08-23:
Yes, please look at our open source CubeTown map repo.  Clone this and look at the pedestrian annotation.  We have fixes in the next release that will out errors on what is missing for random agents and we plan on implementing a notification on the maps themselves on maps that are not open source.

roboserg, 2021-08-24:
> Yes, please look at our open source CubeTown map repo. Clone this and look at the pedestrian annotation. We have fixes in the next release that will out errors on what is missing for random agents and we plan on implementing a notification on the maps themselves on maps that are not open source.

Thanks. Is there a way to know if maps support pedestrians on your Web based SVL sim website?

EricBoiseLGSVL, 2021-08-24:
Sorry, no but we have an issue already covering this. #1549