lgsvl/simulator #1696: sim.reset() does not fully reset a scene (Python API)

roboserg, 2021-08-19:
[Full source code](https://hatebin.com/buqqrmhrcd)

See video for demonstration. I could reproduce the bug even after completely restarting the simulation (press stop in the app, start API sim again)

https://user-images.githubusercontent.com/4758917/129992626-33a3e591-545f-4b74-85ab-9ed304806bc1.mp4

EricBoiseLGSVL, 2021-08-23:
I see.  It was confusing because you pasted in sim.reset under the conditional in the video and when you run it it fails.  Reset should delete all agents in a scene. We will check on this.  How did you get the other two agents?  Was it random agents or add agent?

EricBoiseLGSVL, 2021-08-23:
@revati-naik can you please track this issue? I will check on it also.

roboserg, 2021-08-24:
> I see. It was confusing because you pasted in sim.reset under the conditional in the video and when you run it it fails. Reset should delete all agents in a scene. We will check on this. How did you get the other two agents? Was it random agents or add agent?

1. Towards the end of video I am pasting "sim.load()", not "sim.reset()" just to show that sim breaks in doing so. For the purpose of this github issue please ignore it.
2. As you can see in the video, sim.reset() does not delete the agents, this is the purpose of this github issue.
3. I got the other two agents with the same code you see in the video. Maybe I should have started the video from the start, showing you an empty sim. But basically the code in the video is all there is. Running the code for the first time does "sim.load" and spawns 1x agent as it should. Running the code again, does sim.reset() and spawns a second agent. It doesnt delete the first one.

EricBoiseLGSVL, 2021-08-24:
2) yes, this is a bug, thanks for bringing to our attention. We will test and see how to fix.

EricBoiseLGSVL, 2021-09-16:
We will have this fix in the next release.  Thanks for posting.

closed 2021-09-16 by EricBoiseLGSVL