lgsvl/simulator #1697: add_random_agents for NPC makes my custom NPC stop on the spot.

roboserg, 2021-08-19:
[Full source code](https://hatebin.com/buqqrmhrcd)

Line 33 spawns a named NPC  "Sedan". It drives without problems. As soon as Line 38 is called `sim.add_random_agents(lgsvl.AgentType.NPC)` the "Sedan" stops on the spot. Is it the intended behavior?

I tested to let the Sedan drive for 10+ seconds with a faster speed, it doesn't change anything. As soon as add_random_agents is called "Sedan" stops.

https://user-images.githubusercontent.com/4758917/129993196-d316980d-516e-4371-9dc8-19f4570f9d54.mp4

EricBoiseLGSVL, 2021-08-23:
This might be a bug.  The waypoint behavior should not stop when random npcs are enabled.  @revati-naik please confirm this and make a ticket if it is still happening.

revati-naik, 2021-09-10:
@roboserg We have a fix for this coming in the next release.

closed 2021-09-10 by revati-naik