lgsvl/simulator #1698: LGSVL [SEGMENTAION FAULT] quits immediately after starting

RedBlack888, 2021-08-19:
I tried to run it by following the steps [here](https://www.svlsimulator.com/docs/getting-started/getting-started/) but it didn't work.I chose the nvidia driver.I install ` sudo apt install libvulkan1`, and remove `sudo apt remove mesa-vulkan-drivers`. 

My sever is `Ubuntu 20.04.2 LTS` , `nvidia-smi` as shown:
[image]

Review the Player.log file, as follow
[image]

The log file shows the issue is a bit different from issue #1431 

Could you help to find the bug?

Best.

RedBlack888, 2021-08-19:
As shown in the player.log. I do not find path `/tmp/LGElectronics/SVLSimulator/vulkan_pso_cache.bin`.

RedBlack888, 2021-08-19:
#1479 show the same issue while launching it through remote server. 

After testing locally, the same issue.

EricBoiseLGSVL, 2021-08-23:
Are you running the XRDP service?  If so, it seems that this is a blocker.

umdreamer, 2021-08-24:
> Are you running the XRDP service? If so, it seems that this is a blocker.

I have the same issue. But it is OK in another machine.  I don't know why. The two machines are with the same configuration.

RedBlack888, 2021-08-31:
> Usually a seg fault in unity is from running out of memory. I am guessing the gpu is not loading correctly or driver has an issue. Sorry, we can't help more on your hardware. Maybe @heeen might have some debug tips?

Well, I have solve this issue by intall the right vulkan graph driver for ubuntu 20.04. 
Now, I can start the serve via VNC. 

Thanks.
Best.

EricBoiseLGSVL, 2021-08-31:
Excellent and thanks for posting a solution for others.  Cheers.

closed 2021-08-31 by EricBoiseLGSVL