lgsvl/simulator #1701: Lgsvl Apollo master radar data

tajinder96, 2021-08-19:
Hi, 

I am working with Lgsvl 2021.2 and Apollo-master currently. I was facing the following problems: 

1. radar object detection did not seem to be running - in my understanding this is because the module was not receiving data. I was launching the perception module through dreamview, which links to [the lgsvl specific dag file] (https://github.com/ApolloAuto/apollo/blob/master/modules/perception/production/dag/dag_streaming_perception_lgsvl.dag). I noticed that the topics to the radars do not match the published topic in Lgsvl. If I match the topic name, radar object detection seems to be running. Is this the correct approach?

2. Once I modify the topic name and run the perception module, it seems to crash after some time (see attached image, you can ignore the fusion related messages). Seems like there are tf query related issues now. Is this problem related to the radar sampling frequency, similar to this one? [apollo-issue](https://github.com/ApolloAuto/apollo/issues/8551) Could you please tell me how I can solve this? 

Thank you!
Kind regards, 
Tajinder 

[image]

tajinder96, 2021-08-30:
@hadiTab please let me know if you have any ideas, thanks!

daviduhm, 2021-09-09:
Hi @tajinder96,

You mentioned that the topic of the front radar does not match. What's the topic name that you are using and which sensor configuration for Apollo do you use? I'm asking because we only provide a modular testing sensor config for Apollo master (which is Apollo 6.0 as of today): https://wise.svlsimulator.com/vehicles/profile/73805704-1e46-4eb6-b5f9-ec2244d5951e/edit/configuration/2e9095fa-c9b9-4f3f-8d7d-65fa2bb03921

Modular testing means that the sensor config doesn't include any sensors required for the perception module such as a camera,  a radar, and Lidar because the modular testing sensors directly publish perception and traffic light detection results from simulator.

There seems to be some issues where the perception module is crashing on Apollo side. We provide this modular testing sensor config so that our users can bypass the issue and do not blocked by the perception modules.

You can find more details about modular testing and how to run it with Apollo master in the docs below:
- https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions
- https://www.svlsimulator.com/docs/tutorials/modular-testing