lgsvl/simulator #1702: How to get ego vehicle position in x, y, z coordinates?

dheerajakula, 2021-08-20:
Hi which sensor should I use to get the ego vehicle position in x,y,z co ordinates. I used the gps odometer sensor as mentioned in [https://github.com/lgsvl/simulator/issues/998](https://github.com/lgsvl/simulator/issues/998) but it gives me latitude and longitude globally or relative to origin. But I need the x,y,z unity coordinates like what 3d ground truth sensor gives.

EricBoiseLGSVL, 2021-08-23:
We recommend opening the Unity Editor and running the simulator.  Then, you can create an empty object and see the transform data. We have plans to create a transform tool for the api.  You can also create a sensor that lists all transform data to debug.logs.

dheerajakula, 2021-08-24:
As you said I need to create a new sensor, I found the code for 3d ground truth sensor at [https://github.com/lgsvl/PerceptionSensor3D](https://github.com/lgsvl/PerceptionSensor3D) I think I could tinker this code to accommodate my scenario. 
@EricBoiseLGSVL is there any other other sensor that lists all the transform data to debug.log.

EricBoiseLGSVL, 2021-08-24:
None debug but it would be very easy to modify to get what you want.

dheerajakula, 2021-08-30:
Thanks for the info

closed 2021-08-30 by dheerajakula