lgsvl/simulator #1704: Perception Module Defective in AutowareAI with SVL

marcusvinicius178, 2021-08-20:
Hi @EricBoiseLGSVL @heeen  @lemketron  would you know how to diagnose the origin of the failure in the perception module of AutowareAI while using the LGSVL simulator?

The whole description of my issue is here: https://github.com/Autoware-AI/autoware.ai/issues/2379

I am not expert on Lidar, and its respective filters that feed the occupancy grid map on AutowareAI to determine if there is an obstacle or not in front of the car in rviz.

As the AWF team still didn't answer me this question and always close the issues when I ask this question, I believe that maybe the issue can be a kind of bad alignment with the SVL sensors? I mean set the frequency correctly to send/receive messages or something similar?

It is just a guess, I have no idea why the peception  is not working...I tried numerous times on numerous different powerful machines....and this module rarely works.

A recent attempt is here on youtube video below in CubeTown map. Starts from minute 38 to minute 41:
https://www.youtube.com/watch?v=w6_beZtlLBg

The Ego vehicle hits the NPC car. and the occupancy grid map last a LOT of time! to be generated (you can check I tried about 40 minutes of video to do it!

So it is my last attempt, maybe you can have some solution for me. 
For apollo there is the modular option on perception, so maybe for Autoware_AI it could exist other solution?

On my master thesis I am comparing the ADS (Apollo vs Autoware) to accomplish evasive maneuvers on different scenarios. However I cannot finish the research, if Autoware does not recognize the obstacle and therefore the planner will never work to avoid it, as it is a subsequent step.

EricBoiseLGSVL, 2021-08-24:
Sorry that you are not getting a response from AWF.  I think there is the issue of AutowareAI being older but not sure.  I'm guessing you would need to be able to edit the code.  Have you tried Autoware.Auto?

marcusvinicius178, 2021-08-24:
Yes previously the perception on Autoware.Auto is better however as I described in previous issue there was having issues to launch Autoware.Auto with the new SVL releases. It was working just with the LGSVL 2020.06 however I have checked that this Version is not more available 

Do you know if now the Autoware.Auto is already working with the new SVL releases? Is there new tutorial for this synchronazation? I was trying and it was not working (some issues with the bridge receiving and sending data) ... I mean Autoware.Auto AVP demo was not working

marcusvinicius178, 2021-09-01:
Hi again @EricBoiseLGSVL 
Just to return the solution for the perception module if some other user needs:

The tutorials that I have followed for m AutowareAi in are WRONG! If you launch the costmap_generator through the runtime manager or terminal, the occupancy grid map will not be generated or very very rarely....
To fix this there are 2 options:
1). Using Run-time Manager: on Computing Tab click over app option in costmap_generator, scroll down until you find the Lanelet2 checkbox option at the left bottom. CHECK this box ✔️
2) Inside the docker, using a new terminal launch:
`roslaunch costmap_generator costmap_generator_lanelet2.launch`
Ok now the costmap_generator will generate the occupancy_grid_map
3) Just Pay attention that you must check the ray_ground filter or compare_map_filter on Sensing Tab 

However this will not help as the A* keeps not avoiding and crashing the Obstacle, as described in my issue: https://github.com/Autoware-AI/autoware.ai/issues/2387

If someone fix this A* please let me know

closed 2021-09-01 by marcusvinicius178

EricBoiseLGSVL, 2021-09-01:
Thanks @marcusvinicius178 pinging @cyberphysicalrobotics and @revati-naik on this solution.