lgsvl/simulator #1705: LGSVL crash after ~2hrs of training

ido1Shapira, 2021-08-22:
hey,
i am using SVL SIM for training RL agent using LGSVL GYM library .

and after about 200 episodes, the simulator crushes. it happens several times, and i cant train the agent properly due to does limitations

My environment:
NVIDIA Corporation GP102 [TITAN X]
31.3 GiB RAM
Intel® Core™ i9-10900K CPU @ 3.70GHz × 20 
Ubuntu 18.04

let me know if there is some log that i can provide

hadiTab, 2021-08-23:
Which version of the simulator are you using?
The log is located at `~/.config/unity3d/LGElectronics/SVLSimulator/Player.log`

ido1Shapira, 2021-08-23:
the version we are using is:
svlsimulator-linux64-2021.2

x-special/nautilus-clipboard
copy
file:///home/ido/.config/unity3d/LGElectronics/SVLSimulator/Player.log

EricBoiseLGSVL, 2021-08-24:
This may be a memory leak.  Do you have any more info on the tests you are running?
How many egos, how many npcs, and or peds?
What api commands are you using?

EricBoiseLGSVL, 2021-08-24:
@revati-naik this might relate to the api rest issue

ido1Shapira, 2021-08-24:
@EricBoiseLGSVL - i am using only 1 agent without any NPCs or peds. regarding to the api commends, i am simply trying to train an agent to steer/throttle using RL algorithm. and i mainly use the `on_collusion` call to reset the environment.

another thing is that i upgraded to svlsimulator-linux64-2021.2.2 and still has the same issue
here is the log file for the new version crush:
[Player.log](https://github.com/lgsvl/simulator/files/7036891/Player.log)

can you elaborate on the memory leak issue?

ido1Shapira, 2021-09-05:
@EricBoiseLGSVL any news on this subject?
and to you have estimate time for 2021.3 release date?

EricBoiseLGSVL, 2021-09-08:
We are on code freeze this week. :) We need rounds of testing and then docs.  Should be in a few weeks

yeehsienq, 2021-11-16:
Hi, I'm training an RL agent in the Highway101GLE environment, and there is a Segmentation fault after a few thousand episodes of training. I also suspect that the reason for this is a memory leak.

My environment:
Ubuntu 18.04.5 LTS
Processor: Intel® Core™ i7-3770 CPU @ 3.40GHz × 8
Graphics: NVIDIA GeForce RTX 3060/PCIe/SSE2

Any help is appreciated and let me know if you need more information!

EricBoiseLGSVL, 2021-11-16:
it looks like the socket is closed when trying to send the test report.
```
[ANMGR] Initializing with TestReportId:d09f1269-58b3-48f3-a4a2-bbde6ab05990
16/11/2021 4:12:08 PM|Fatal|<>c__DisplayClass174_0.<startReceiving>b__2|WebSocketSharp.WebSocketException: The header part of a frame could not be read.
                              at WebSocketSharp.WebSocketFrame.processHeader (System.Byte[] header) [0x00010] in <fc1f0434ffca46a29e99654e48881787>:0 
                              at WebSocketSharp.WebSocketFrame+<>c__DisplayClass73_0.<readHeaderAsync>b__0 (System.Byte[] bytes) [0x00000] in <fc1f0434ffca46a29e99654e48881787>:0 
                              at WebSocketSharp.Ext+<>c__DisplayClass57_0.<ReadBytesAsync>b__0 (System.IAsyncResult ar) [0x00078] in <fc1f0434ffca46a29e99654e48881787>:0 
```

Can you disable the test report feature and see if you still get the crash?

EricBoiseLGSVL, 2021-11-16:
It also seems you are loading the cyber bridge multiple times.  Can you post your sensors used and any changes you have made to source?  How are you resetting the simulation.  Please step through how you are running simulations because memory leaks are the hardest to trouble shoot.  It may even be an issue with the Unity engine.

yeehsienq, 2021-11-17:
I still get the segmentation error when the test report feature is disabled.
[Player.log](https://github.com/lgsvl/simulator/files/7551714/Player.log)

I'm actually not using the cyber bridge, but just using the Python API. But if loading the cyber bridge is causing the problem, how can we disable that? For sensors, I currently use one main RGB camera and 3 depth sensor cameras. 

In my training script, I'll reset the simulation by calling env.reset() either after a fixed number of steps have passed, or when the vehicle experiences a collision.

EricBoiseLGSVL, 2021-11-17:
if you are not using the bridge then yes, use a vehicle that has no bridge.  This might be the issue.

yeehsienq, 2021-11-17:
May I ask, how do we remove the bridge from the vehicle?

EricBoiseLGSVL, 2021-11-17:
in the vehicle config in the webui.
[image]