lgsvl/simulator #1706: WebUI fails to upload custom vehicle asset;  assetUploader.errors.(undefined)

mozhoku, 2021-08-23:
Hello SVL team,

   I have two custom vehicle bundles that I built and uploaded before the release of SVL 2021.2. Assets were built with SVL 2021.1.1 and were working fine.
After the new release I was unable to upload one of the vehicles, while other one was okay. I tried uploading the vehicle bundles built from v2021.1, v2021.1.1, v2021.2.1, v2021.2.2 and same issue persisted.
[image]

   I've checked the bundle contents for the failing one and everything was okay. Later on I added the failed bundle to simulator environment by manually installing bundles and editing data.db file located in \AppData\LocalLow\LGElectronics as said in here  https://github.com/lgsvl/simulator/issues/1692#issuecomment-901497712. There was no problem with the vehicle bundle, it worked in simulator. But I'm still unable to upload it to the SVL website.

   I know similar issues opened before. Why does the website accept one bundle and gives error with the other one? Is there anything I can do to make the failing bundle upload like the other one?

I've tried making bundle from different SVL versions, remaking vehicle asset from scratch, changing bundle name, using edited 3D model.

Similar issues: https://github.com/lgsvl/simulator/issues/1683 , https://github.com/lgsvl/simulator/issues/1585

EricBoiseLGSVL, 2021-08-24:
I think you have issues with conflicts in the database.  @eagafonovlge @thisisdavidfrank can you take a look when you have time?
@mozhoku I would remove all versions of the vehicle you are trying to upload.  Unlink the cluster. Relink the cluster and upload.  When it asks if you want to update the older version or make a new, choose make a new asset. Sorry about the issues and I hope this helps

mozhoku, 2021-08-24:
@EricBoiseLGSVL Thanks for reply. 
Today I've removed vehicle from WebUI and unlinked all clusters as you suggested. Then tried uploading bundles built from SVL 2021v1.1 and 2021v2.2 again. It gives the same error.
[image]

alperenbayraktar, 2021-08-24:
@EricBoiseLGSVL Just relinked and deleted my uploaded ego car, tried to re-upload it and this error shows up regardless of the simulator versions I use(my custom one and the release 2021.2). I tried to change my vehicle name and upload it again but it doesn't help.

burnout2k, 2021-08-29:
Have the same problem. Cant add a new vehicle. When I deleted my old one and tried to add the same vehicle file I got a duplication error as well. Something is really messed up when it comes to your database and adding vehicles...

EricBoiseLGSVL, 2021-08-31:
Does this happen if you create a new vehicle? Yes, sorry about the issues.  We have been implementing many changes for the next release to reduce these issues.

thisisdavidfrank, 2021-08-31:
@alperenbayraktar, @mozhoku, and @burnout2k: Sorry about the issues that you are facing. We acknowledge that the current web interface has several bugs related to asset uploads. The bugs are mostly in the areas of version incompatibility checks between the various Simulator releases. We have fixed every one of these issues that we have seen so far, and you should have the latest version of the web interface at the end of September along with more advanced features. 

In the meantime, can you let me know what you would like from us to resolve your problem? I'll be working with some developers this week to figure out what exactly is breaking in your asset bundles. Once we have that answer, we can help resolve it as well. Is here anything else that you might need?

burnout2k, 2021-09-01:
@alperenbayraktar I tested with some other models and was able to get them working with no problem.
I think the problem might be the file size or the polygon count.
My car is around 160 MB and pretty high poly (800k+)
here is the link to the successful build that I am not able to add through the web interface
https://drive.google.com/file/d/1nz0GrevJiOSXFJxXbq7O5PcxOa6jBNxV/view?usp=sharing
maybe you can check it out
Also, another thing, when it comes to the models that do work, they are not compatible with the latest build, only with versions 2021.2 and 2021.2.1 which is weird (I am using Unity 2020.3.3f1)

EricBoiseLGSVL, 2021-09-01:
Ah, that makes sense.  I don't think we allow that big of a file.  What version of simulator did you build the asset bundles?

alperenbayraktar, 2021-09-01:
@burnout2k  Well, my ego vehicle is also 134mb.

@thisisdavidfrank It would be great if you could provide a hotfix this week, the site should get rid of the upload errors such as:
-undefined error when uploading
-duplicated asset error when uploading which the user already deleted 
-Sensors not showing up on the vehicle sensor config #1714 
-bridges not showing the autopilot type #1713

burnout2k, 2021-09-01:
@EricBoiseLGSVL Sorry, had not cloned the latest version, my mistake...

ertauren, 2021-09-02:
> 
> 
> Ah, that makes sense. I don't think we allow that big of a file. What version of simulator did you build the asset bundles?
@EricBoiseLGSVL 
Hi, what is the allowed file size for vehicle?
https://github.com/lgsvl/simulator/issues/1683
this issue of mine probably have same problem. Because my vehicle is 194mb. I will also try with latest version soon.

mozhoku, 2021-09-02:
Thanks for the help @EricBoiseLGSVL @burnout2k @thisisdavidfrank . 
Didn't know there was a size limit for uploading. 
Problem was my bundle size. I've succesfully uploaded my vehicle with a smaller bundle. 🥳 

WebUI should say "your bundle too big" instead of throwing undefined error. Please specify upload limit for the asset files so people can prepare their assets accordingly. 🙏

EricBoiseLGSVL, 2021-09-02:
@thisisdavidfrank Can you post the maximum bundle size here so we can point users to this issue please?

closed 2021-09-02 by EricBoiseLGSVL