lgsvl/simulator #1707: Error with bridge.sh using Apollo 5.0

WhoAmI-kai, 2021-08-24:
I've been trying to run the simulator with Apollo 5.0 and when I execute bridge.sh, I get the following error message:
`scripts/bridge.sh: line 6: bazel-bin/cyber/bridge/cyber_bridge: No such file or directory`

Any ideas on what I'm doing wrong? One error message I also see in the previous step (I don't know if it's related or not) is that when I execute `bootstrap.sh`, it says:
```
nohup: appending output to 'nohup.out'
nohup: appending output to 'nohup.out'
Could not launch module monitor. Is it already built?
nohup: appending output to 'nohup.out'
Launched module dreamview.
Failed to start Dreamview. Please check /apollo/data/log or /apollo/data/core for more information
```

EricBoiseLGSVL, 2021-08-24:
@revati-naik @hadiTab any idea?

revati-naik, 2021-08-24:
@WhoAmI-kai Please follow the instructions provided [here](https://www.svlsimulator.com/docs/archive/2020.06/apollo5-0-instructions/) to build the Apollo 5.0 container. After you enter the container, please build Apollo using `./apollo.sh build_opt_gpu
` before running `bootstrap.sh`

WhoAmI-kai, 2021-08-25:
I have used` ./apollo.sh build_opt_gpu` before running `bootstrap.sh`, but only the first time it can run。
When I restart the container or restart the machine and ran `bootstrap.sh`, I get the following error message:
```
nohup: appending output to 'nohup.out'
nohup: appending output to 'nohup.out'
Could not launch module monitor. Is it already built?
nohup: appending output to 'nohup.out'
Launched module dreamview.
Failed to start Dreamview. Please check /apollo/data/log or /apollo/data/core for more information
```
Then, when I executed `bridge.sh`,  I get the following error message:
`scripts/bridge.sh: line 6: bazel-bin/cyber/bridge/cyber_bridge: No such file or directory`

WhoAmI-kai, 2021-08-30:
[image]

revati-naik, 2021-09-07:
@WhoAmI-kai As fasr as what I have noticed is that while using a dev container, we need to build the container everytime we enter the container. We test with the runtime container and do not see this issue. If it suits your need, woudl you mind testing with the runtime container. Could you please follow the steps given [here](https://github.com/lgsvl/apollo-5.0)?

AdelinaK, 2022-02-26:
> @WhoAmI-kai As fasr as what I have noticed is that while using a dev container, we need to build the container everytime we enter the container. We test with the runtime container and do not see [...]

I tried to build the simulator and I've got the same problem. In the end which of the containers should be used?  Runtime container or normal container? The documentation is quite confusing too. I'm using Ubuntu 20.04 and after I start a simulation with an Apollo 5.0 controller the simulator crashes. Maybe some updates on the documentation and explaining how the whole sistem works will help. Thank you!