lgsvl/simulator #1711: traffic lights editing in VSE editor

nijilong1224, 2021-08-24:
Hi all:
How to edit traffic lights behavior in windows VSE, from the Python API guide, it can change its behavior by updating control policy: "trigger=50;green=1;yellow=1.5;red=2;loop", while in VSE I only can update policy in this manner.
[image]
does that mean each state will wait for several secs specified in wait field? Also if I edit one traffic light in junctions, other lights policy will changed automatically.
[image]
Seems it's not a correct policy.

Thanks.

EricBoiseLGSVL, 2021-08-24:
Yes, this is the functionality that we support as of now.  We are working on supporting more but it is not ready yet.  What do you think would be the correct policy?  Any feedback is great and helps us create better features.  @KarByc

KarByc, 2021-08-24:
Hello @nijilong1224 ,

Policy "trigger=50;green=1;yellow=1.5;red=2;loop" can be achieved by replacing the PythonAPI lights color command with a pair of "state" and "wait" commands in the VSE. Please try setting it like below:
[image]

If lights do not behave as expected please include videos of the expected behavior (set with the PythonAPI) and the actual behavior (set with the VSE) with their Python API code and VSE setting screenshot. That will help me to reproduce the issue.
Thanks!

nijilong1224, 2021-08-25:
Hi @EricBoiseLGSVL and @KarByc

Thanks for your quick reply.
@KarByc, The lights behavior through Python API is right.
Since, I could not run VES template in windows simulator, so I decide to generate VSE template file, parse the file in python, then run the scenario through APIs.
my concern is if I edit only one traffic light policy in VSE, other lights in same junction will have some policy value automatically, is that correct?
[image]
editing the middle light policy with "trigger=20;green=20;yellow=3;red=20;loop"
[image]
check the right light policy
[image]
[image]
check the left 2 lights policy
[image]
check middle light policy again, the loop tag is missing.
[image]
the final VSE template file contents corresponding to lights, no "loop" action.
how to distinguish the controllable type, I only want to retrieve traffic lights info.

BTW, could I use Playback function to check traffic lights behavior in VSE editor.

Thanks a lot!
Best Regards

nijilong1224, 2021-09-03:
Hello @KarByc 

It's great, thank you very much!

Best Regards

closed 2021-09-05 by nijilong1224