lgsvl/simulator #1712: Apollo cannot autodrive the car in simulator

umdreamer, 2021-08-24:
I have installed the svlsimualtor 2021.2, and Apollo 6.0 in Ubuntu 16.04 with Nvidia driver 450 with CUDA 11. Everything seems OK according to the documentation: https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/.

But in the svl and apollo simulation, the car stopped, and could not go automatically. It seems that I cannot designate the starting point and ending point of the car. But how to do it in Apollo Dreamview pannel? It didn't work after I clicked the mouse.
[image]

EricBoiseLGSVL, 2021-08-24:
@revati-naik @lemketron ?

revati-naik, 2021-08-24:
@umdreamer Please make sure you turn on the modules in Dremaview under the tab "Module Controller". Follow the instructions given under [Launching Apollo alongside the Simulator](https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/). 

Also, please make sure you give a valid destination point. For example, start with placing a destination point along the same lane-line (green line) as the ego vehicle.

umdreamer, 2021-08-25:
@revati-naik Yes, I have done according to the documentation. Here are some snapshots of my program.
[image]
[image]
[image]
There are two points I placed. Are they correct?
[image]

lemketron, 2021-08-25:
> There are two points I placed. Are they correct?

Please click `Remove All Points` and then click only once to set a destination directly in front of the ego vehicle.

As noted in the Apollo Dreamview documentation, you should see a red line indicating the route (after you click to set a valid destination and then click "Send Routing Request" if the `Routing` module is enabled) and then (if the `Planning` module is enabled) you should also see a "green thick curvy band [which] indicates the planned trajectory". At that point, if there is no stop fence in front of the ego vehicle, then if the `Control` module is enabled, the ego vehicle should start driving. Sometimes hitting the up arrow on your keyboard to nudge the car a bit can also help (and will ensure that the vehicle IS able to move).

umdreamer, 2021-08-25:
@lemketron I made it. It works now. Thank you very much.