lgsvl/simulator #1713: Bridge Autopilot Type Doesn't Show Up

alperenbayraktar, 2021-08-24:
[image]
I have uploaded a custom ROS bridge but it doesn't show the Autoware options on the Simulation options tab.
[image]

EricBoiseLGSVL, 2021-08-24:
Adding @mark-gerow-lge to confirm custom bridge has been set correctly. Thanks

mark-gerow-lge, 2021-08-24:
Can you show me what your BridgeNameAttribute tag on your IBridgeFactory implementation?

alperenbayraktar, 2021-08-25:
> 
> 
> Can you show me what your BridgeNameAttribute tag on your IBridgeFactory implementation?

@mark-gerow-lge  It's name and type are both "CustomRos"

[image]

Maestroxr, 2021-08-25:
This also happens with the original ROS2 bridge that is on the site's store by default.
The Autoware options are grayed out even if the built-in bridge is used and this just happened after hanging the bridge and changing it back, so things that worked until recently stopped working now because I can't revert the setting and be able to save the simulation setting.

alperenbayraktar, 2021-09-06:
@EricBoiseLGSVL Updating the asset doesn't fix the issue, it is still green and autopilots doesn't show

alperenbayraktar, 2021-09-08:
> 
> 
> @mark-gerow-lge can you look at this again please?
> @alperenbayraktar Have you purged all bridges and uploaded the new one with the `[BridgeName("CustomRos", "ROS")]` changes?

Yes, as I said, I've tried with both ROS and CustomRos types before too. It has ROS as type, double-checked it from the manifest. Removed the configs that use the bridge from my library and then the bridge itself, uploaded it again. The bridge doesn't show up on the "Public" tab now and I just signed up a second account and shared the bridge because it doesn't show up on the vehicle config when this happens. The second account's vehicle config is also green and autopilots don't appear.