lgsvl/simulator #1722: Simulation Execution Questions

yanghyunseok3, 2021-08-30:
Hello.
When I pressed 'Run simulation', it ended as soon as it popped up in the SVLSimulator window.
It returns to the screen before running the simulator again.
I don't know why this is happening all of a sudden even though I didn't install anything special. I wonder if there are any cases of this happening.
[image]
[image]

yanghyunseok3, 2021-08-31:
When I ran it again today, it suddenly worked normally. I'm sorry.

EricBoiseLGSVL, 2021-08-31:
No worries, glad it is working.  Next time, please post and review the player.log from simulator.  It will log errors and show you what could be the issue.

yanghyunseok3, 2021-08-31:
For now, however, after selecting the vehicle, running the simulator using a set of 'sensor configuration' designed for my experimental purposes will terminate the simulation at the same time as running for two seconds. I don't know why. I use 2021.2 version. The keyboard control works normally. Two weeks ago, after checking the normal operation using the sensor configuration made by myself, there was no specific modification. Which part should I check?
[image]
[image]
[image]

yanghyunseok3, 2021-08-31:
It runs too instantaneously and then shuts down, so I cannot see the console screen indicating which part is the error...

EricBoiseLGSVL, 2021-08-31:
No there is a player.log in the AppData folder for SVLSimulator. See [here](https://docs.unity3d.com/ScriptReference/Application-persistentDataPath.html).

yanghyunseok3, 2021-08-31:
I found the 'player.log' and opened it. And then, I found the changed things when I launched the simulator. Here is the recorded log.

loader status: Idle->Loading 

[LOADER] Update simulation status Idle -> Loading
[CONN] Updated simulation 53999d7c-187a-4680-9b40-3a5f3af8d73f with status:Starting msg:
[CONN] POST /api/v1/simulations/53999d7c-187a-4680-9b40-3a5f3af8d73f/status
Initialized the Simulation Network data.

found manifest at 24ad2e98-6732-4555-99b8-a8a526218abd/manifest.json

Bridge ROS2 already registered, ignoring duplicate(?) plugin

found manifest at 3614201c-958e-4be7-9db9-bd0e3fdc7116/manifest.json

found manifest at 9f658040-e590-410b-abdd-38797987c1a2/manifest.json

found manifest at 309ad091-c74e-4074-9e0b-ed068f6c90ea/manifest.json

found manifest at b318f5c9-1aac-41f5-b597-080a7713ad7d/manifest.json

found manifest at 3614201c-958e-4be7-9db9-bd0e3fdc7116/manifest.json

found manifest at 3614201c-958e-4be7-9db9-bd0e3fdc7116/manifest.json

found manifest at 3614201c-958e-4be7-9db9-bd0e3fdc7116/manifest.json

found manifest at 5ae06173-0ec5-45dc-95c3-1de27cb2869d/manifest.json

found manifest at 9f658040-e590-410b-abdd-38797987c1a2/manifest.json

found manifest at 5f931d55-b797-40d5-b807-7a5892e7c0bd/manifest.json

found manifest at 8f9e9e34-0220-40aa-b85b-a6f0c45d7ed1/manifest.json

found manifest at 3d765dbb-6bb2-4f77-80f0-88bec7eeaf86/manifest.json

found manifest at 391217cd-3ab1-4544-90ac-ea91c5f8f08f/manifest.json

found manifest at 3614201c-958e-4be7-9db9-bd0e3fdc7116/manifest.json

found manifest at dc5c631c-2f58-4db1-9fe6-06bc4e801bf4/manifest.json

found manifest at 3614201c-958e-4be7-9db9-bd0e3fdc7116/manifest.json

All Downloads Complete

loader status: Loading->Starting 

[LOADER] Update simulation status Loading -> Starting
[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

Unloading 2 Unused Serialized files (Serialized files now loaded: 3)
UnloadTime: 1.034445 ms
Initialized the Simulation scene for the network manager

found manifest at 24ad2e98-6732-4555-99b8-a8a526218abd/manifest.json

Bridge ROS2 already registered, ignoring duplicate(?) plugin

Failed to start 'yang_test' simulation

Sentry exception condition not valid

NullReferenceException
  at (wrapper managed-to-native) UnityEngine.Component.get_gameObject(UnityEngine.Component)
  at SensorsController.InstantiateSensors (Simulator.Web.SensorData[] sensors) [0x00104] in <06911aa1667a42f9ab2a8b7dd4beae4a>:0 
  at SensorsController.SetupSensors (Simulator.Web.SensorData[] sensors) [0x00016] in <06911aa1667a42f9ab2a8b7dd4beae4a>:0 
  at AgentManager.SpawnAgent (Simulator.AgentConfig config) [0x001c7] in <06911aa1667a42f9ab2a8b7dd4beae4a>:0 
[... 116 lines cut ...]

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 5256.
Total: 28.967697 ms (FindLiveObjects: 0.445012 ms CreateObjectMapping: 0.271562 ms MarkObjects: 28.227008 ms  DeleteObjects: 0.022750 ms)

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] Updated simulation 53999d7c-187a-4680-9b40-3a5f3af8d73f with status:Idle msg:
[CONN] POST /api/v1/simulations/53999d7c-187a-4680-9b40-3a5f3af8d73f/status
[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

I wonder which part is wrong and what can I do..? I'm sorry.

EricBoiseLGSVL, 2021-08-31:
Be sure to be using the latest 2021.2.2 release
It looks like you have an issue with the sensor.  Please remove all local cached sensors and simulations in the main menu top right.  Then unlink and relink the cluster.
Remove your assets and add back from the library.
Then remake the simulation.
Sorry about the issues.  We have have some asset compatibility issues with 2021.2 and we have fixed most of them in the up coming release 2021.3.