lgsvl/simulator #1723: What does Utility.SqrDistanceToSegment use for please?

GooVincent, 2021-08-30:
I am new in lgsvl simulator, and can't figure out what Utility.SqrDistanceToSegment use for? and how does it work? Please anyone could help me.

EricBoiseLGSVL, 2021-08-31:
@GooVincent SqrMagnitude is a more efficient way to calculate distance than using Vector3.Distance.  This is used to measure distance from a point to a line segment. It is used in StopLineSensor, ApolloMapImporter, Lanelet2Importer and MapManager.
See unity docs, it's a good explaination [here](https://docs.unity3d.com/ScriptReference/Vector3-sqrMagnitude.html).

GooVincent, 2021-08-31:
Thanks. I am looking for a way to evaluate if the vehicle could follow the lane.  I viewed the code of LGSVL simulator follows the lane. It selects the closest lane and generates a list of waypoints. But how to make sure it matches the LKA standard of NCAP. That quantizes the distance between the vehicle and lane line.

EricBoiseLGSVL, 2021-08-31:
Sure, but our lane following is for NPCs.  They don't use wheel colliders but move the rigidbody along the waypoints.  Why are you testing the lane following on the NPCs not the AD stack algorithms for the EGO? Also, the sqr magnitude method is just a common measuring algorithm, how does that relate to the standard? Are you trying to make a sensor that measures this for the EGO?

GooVincent, 2021-08-31:
Well, I just wanna learn how the simulator implemented lane following. As I need to evaluate the algorithm effectiveness which runs in the EGO vehicle. I wanna know how to use LGSVL to test if a certain algorithm like lane following works or not.

EricBoiseLGSVL, 2021-08-31:
But we don't provide lane following for EGO.  The simulator is for testing the AD stack algorithms.  I recommend creating a sensor for the ego that when controlled by Apollo or Autoware, outputs the measurements and you can even add events to pass fail the simulation when LF is outside the spec.  Make sense?

GooVincent, 2021-09-01:
Thanks for your reply. 
Please any docs for how to make a customer sensor. I have cloned the Comfort Sensor and build it successfully as a new sensor. But regarding my own new customer sensor, its attribute should be new to the simulator, for instance, the distance from the ego vehicle to the lane line. How to make the kind of new attribute to be sensitive.

EricBoiseLGSVL, 2021-09-01:
No problem. Sensor plugin docs [here](https://www.svlsimulator.com/docs/plugins/sensor-plugins/).
You can also look at our other open source sensors. Be sure the name of the folder/class and prefab are the same and include the new data class, e.g., ComfortData.cs