lgsvl/simulator #1724: Camera data visualization from Apollo cyber_visulaizer

zyan1234, 2021-08-31:
Hi,

I am trying to run SVL simulator along side with Apollo stack. The simulator and AD stack are run on different machines.

I have created a simulation with Lincoln2017MKZ vehicle, BorregasAve map, and with Apollo 5.0 sensor configuration. The Apollo 5.0 sensor configuration has main camera sensor in it.
The simulator gets connected to Apollo stack with the Cyber on the specified IP of the Apollo machine.
The bridge is functioning properly, data is getting flowed through cyber bridge to Apollo stack, dreamview is working properly, vehicle moves well on giving the destination.

But to visualize the data through cyber_visualizer I can visualize the point cloud data but not the camera data.

[image]

I tried with both front_12mm and front_6mm, but no image is getting displayed
Please let me know how can I visualize the camera data.

Thanks

EricBoiseLGSVL, 2021-08-31:
are you using the full topic specified in the sensor?
```/apollo/sensor/camera/front_6mm/image/compressed```
What version of simulator and config are you using? 5.0 modular does not have a main camera sensor

zyan1234, 2021-09-01:
@EricBoiseLGSVL  
Yes I am using the full topic specified in the sensor **/apollo/sensor/camera/front_6mm/image/compressed**
The version of the simulator used is 2021.1
I again checked with the apollo 5.0 sensor configuration, it does have a main camera sensor
[image]

I meant the camera sensor from here, let me know if this is incorrect
I have added this sensor config to the vehicle and trying to visualize the camera data with topic **/apollo/sensor/camera/front_6mm/image/compressed** from cyber_visualizer

Help me to visualize camera data from cyber_visualizer