lgsvl/simulator #1725: Abnormal Scene display in Unity

ideasplus, 2021-08-31:
Hi Team,

I want to modity the specific map to add some 3D cubes,  so I open the map in Unity, edit and build it according to official document.  

However, when I run simulation from the modified map, the scene display is abnormal as shown in the following. Why does this happen and how can I solve this problem? 
The second question is when I add a 3D cube to the map,  what attributes should I add to it? and how can I change the size of the cube?

Note that I'm using lgsvl 2020.06 and Unity 2019.3.15f1.

Here is the screenshot when I open SingleLineRoad in Unity 
[image]

Here is the screenshot when I add a cube to  SingleLineRoad map
[image]

Here is the screenshot when I run simulation from the modified SingleLineRoad map
[image]

Here is the screenshot when I open CubeTown in Unity
[image]

Here is the screenshot when I run simulation from the modified CubeTown map
[image]

EricBoiseLGSVL, 2021-08-31:
Why are you using simulator 2020.06? This is very old.
What tag do you have the cubetown repo at?
Did you have git lfs running when you cloned cubetown?

ideasplus, 2021-09-01:
> Why are you using simulator 2020.06? This is very old.
> What tag do you have the cubetown repo at? Be sure it is 2020.06
> Did you have git lfs running when you cloned cubetown?

I prefer to use the simulator 2020.06 because I've been using this version for a long time.
Thanks for your reminding. I have solved this problem.

Meanwhile, I have a new little question, How can I switch this view in simulator like this?
[image]

EricBoiseLGSVL, 2021-09-01:
Can you please post the solution to help others?
This is the driver view camera mode that is available in the latest simulator.

ideasplus, 2021-09-02:
> Can you please post the solution to help others?
> This is the driver view camera mode that is available in the latest simulator.

Yes, I solve the problem by switch the version of the map to 2020.06

Thanks for your help.

closed 2021-09-02 by EricBoiseLGSVL