lgsvl/simulator #1726: After the traffic crash, the npcs do not disappear

gizem-buss, 2021-08-31:
Hello, I am creating npc's with python api and I provide the movements of these npc's with the follow closest lane. After the traffic crash, the npcs do not disappear after a certain period of time. How can I provide?

EricBoiseLGSVL, 2021-08-31:
You can have a callback on collision and sim.reset

gizem-buss, 2021-09-01:
thx,@EricBoiseLGSVL . When I use sim.reset, does it disappear in all scenarios I created with the python api? What I want to do in python api is that when two npc collide or traffic jam, those vehicles are destroyed.

EricBoiseLGSVL, 2021-09-01:
You need to change the simulator source to despawn on collision or on trigger. It's the same code that calls the ApiManager to add the call back

closed 2021-10-14 by gizem-buss