lgsvl/simulator #1729: Request for source code of DallaraIL15 ego

ByeongjunShin, 2021-09-01:
To lgsvl

Hello, I am Byeongjun Shin, an undergraduate student, participating in Indy Autonomous Challenge(IAC) as an intern of KAIST Lab under a professor. For the challenge, our team explicitly uses this simulator now, so I really appreciate for giving such a nice tool to us.
The purpose of this issue is to know whether getting the source code of DallaraIL15 ego is available as we can get the open source of the Jaguar2015XE in github(https://github.com/lgsvl/Jaguar2015XE).

Since we want to get DallaraIL15s with different colors(only red color in the simulator now) for perception of the autonomous car, which will be used for getting training data set, we hope to get the open source of DallaraIL15.  Changing the color of Jaguar is available using building Assetbundles after modifying colors in unity; on the other hand, it is really hard to change the color of DallaraIL15.
To sum up, my question is that
May I get the source code for DallaraIL15 ego to be used for generating various-colored DallaraIL15? in the same format as we can get by git cloning from https://github.com/lgsvl/Jaguar2015XE which could be put in simulator/Assets/External/Vehicles and built for AssetBundles.

Thank you :)

EricBoiseLGSVL, 2021-09-01:
@ByeongjunShin Welcome to SVL Simulator.  Glad you are liking it.
Sorry, we cannot open source the DallaraIL15.  This asset was modified from a purchase and the license prohibits us from open sourcing it.
Still, there are ways to alter the open source simulator and API code to achieve this.
If you need just random colors you can simply alter AgentManager.cs SpawnAgent() method. I put this in line 78 but anywhere before segmentation is fine
```
var renderers = go.GetComponentsInChildren<Renderer>();
foreach (var renderer in renderers)
{
    foreach (var mat in renderer.materials)
    {
        if (mat.name.Contains("Body"))
        {
            mat.SetColor("_BaseColor", UnityEngine.Random.ColorHSV(0f, 1f, 1f, 1f, 0.5f, 1f));
        }
    }
}
```
this gives you this after spawning the vehicle
[image]

Keep in mind that you will need to create a new binary build of simulator.

If you need specific colors you will need to edit the Python API source and the API commands in simulator.  Look at how NPCs have the option to set a color as an example.

Hope this helps.

ByeongjunShin, 2021-09-02:
Dear Eric

Hello, Eric. I got your email and changed the code right away, and finally,
I changed the color of DallaraIL15s!

I am really pleased to make this work and thank you so much!

Sincerely,

Byeongjun Shin

p.s. I attached my result! Thank you!

EricBoiseLGSVL, 2021-09-02:
Glad to help. Good luck in the Indy Autonomous Challenge(IAC)!

closed 2021-09-02 by EricBoiseLGSVL