lgsvl/simulator #1730: How to modify the appearance of obstacles?

ideasplus, 2021-09-02:
Hi Team,

I want to modify the appearance of  specific obstacles and **run simulation with them**. I couldn't find relevant tutorial on it. Could you provide me some help to achieve this goal?

Firstly, I want to modify the NPC such as adding a patch at specific position on car body. The effect is as follows.

https://user-images.githubusercontent.com/24294571/131774400-ec7c489c-3456-402e-b069-9948b3fe4ccf.mp4

Secondly, I want to modify the stop sign in the CubeTown map such as adding some noise on it.
[image]

EricBoiseLGSVL, 2021-09-02:
@ideasplus 
Since NPCs and CubeTown are open source, you can make any changes you need.

- For the NPC texture, I recommend looking at the MKZ vehicle source for the decal projector unity component in the DecalLeft and DecalRight game objects in the vehicle hierarchy.  That way you don't need to adjust the NPC texture.  You can even add scripts to the NPC or source to modify the decal position at runtime.

- I recommend changing the texture for the stop sign.  Look to the material that uses an HDRP lit shader.  There you can add normal maps, change the texture, change specular settings, etc. You could even create and replace the shader with more options for visuals that you may need.

I also recommend looking into some Unity tutorials to see how HDRP, shaders, decal projectors work. Unity's documentation is a great resource into using the engine.

Hope this helps

ideasplus, 2021-09-03:
@EricBoiseLGSVL 
Thanks for your feedback. 

I'm sorry that I didn't clarify the second requirement.  I want to replace the original stop sign with my modified stop sign.  Could you give me some suggestions on how to achieve it?

EricBoiseLGSVL, 2021-09-03:
Glad to help and no worries.
- Drop the new mesh in the scene and make a prefab.
- Align the sign with the one you want to replace, remove the old and save the scene.
- Build the bundle.  Be sure all the assets you change or add are in the CubeTown repo folders.
Hope this helps

tianyunzhe, 2021-11-19:
> 
> 
> @EricBoiseLGSVL Thanks for your feedback.
> 
> I'm sorry that I didn't clarify the second requirement. I want to replace the original stop sign with my modified stop sign. Could you give me some suggestions on how to achieve it?

Hi, I am a beginer. I meet the same task. Could you provide me some detailed steps to achieve this goal? Many thanks!

EricBoiseLGSVL, 2021-11-19:
[image]

1) Place the meshes/materials/textures/prefab in the CubeTown folder so when you build the map, it bundles these with it
2) Place the prefab of your new road sign in the CubeTown scene, delete or disable signs you don't need.
3) Save the scene
4) Build the map bundle and upload to webui.  Be sure to name it different so know it's the altered one.

Good luck, you got this.  Be sure to check out any Unity Editor videos if you are confused on how to use the Editor.

closed 2021-11-19 by EricBoiseLGSVL