lgsvl/simulator #1732: Map Annotation : Point Cloud Map can't match with opendrive

SampsonM8, 2021-09-03:


EricBoiseLGSVL, 2021-09-03:
Can you please post some images so we can see the issue?
Also, what version of simulator are you using?
How did you make an open drive map? Do you set the MapOrigin to match the Point Cloud origin?
More information will help answer this issue.

SampsonM8, 2021-09-06:
Hi, @EricBoiseLGSVL .Sorry about replying you after 2 days.1
**1**. I am using master branch and commit version is 
commit 90c47ce664da6f993da5e7f463213e6f0c667272 (HEAD -> master, origin/master, origin/HEAD)
Author: eric.boise <eric.boise@lge.com>
Date:   Fri Jul 2 09:01:41 2021 -0700

    [FEATURE MAPS] [-] Annotation size
2. The way i made map is that i drived a car with GPS , lane detection devices in the middle of lane . I recorded gps data and lane width from lane detection camera device. With these road information,  I calculated geometry reference points and suggeset the road as a plane. The PointCloud and MapOrigin were set. And the map annotation displayed that they matched with each other.
3. The pic shows as following

SampsonM8, 2021-09-06:
[image]

SampsonM8, 2021-09-06:
As shown in the above pic, two lanes show up in the left road .
I made that the two lanes are right in opendrive map file. And in point cloud map, right direction is same as the above picture right direction. 
I think that lgsvl is left hand coordinate and the Y axes is downward. So the left direction of picture is actually right direction of lgsvl simulator.

So i personally modified **GetNormalDir Function** codes in OpenDRIVEMapImporter.cs file. 
I repeated origin code that 
return normalDir.normalized * (isLeft ? 1 : -1);
as 
return normalDir.normalized * (isLeft ? -1 : +1);

So the direction is exchanged. And the road shows up in the right direction of point cloud. 
But i can't export the opendrive file successfully. 
I clicked the export buttion and the unity3d programme broken down immediately. There are not any tips in console window

SampsonM8, 2021-09-06:
May be my solution is totally wrong. 
I would be very happy to receive  your replay @@EricBoiseLGSVL.

SampsonM8, 2021-09-06:
[image]
The pic is about the whole map.

SampsonM8, 2021-09-06:
Hi, @EricBoiseLGSVL 
I solved the problem. 
In MapLaneSection.cs File, i should change the judgement of  neighbour lane direction to generate "correct" neighbour lane data. I modified **SetLaneData**() method like this:

if (isSameDirection) // same direction
                    {
                        if (cross < 0) // otherLane is left of lane
                        {
                            if (dist < minDistLeft) // closest lane left of lane is otherLane
                            {
                                minDistLeft = dist;
                                lane.leftLaneForward = otherLane;
                            }
                        }
                        else if (cross > 0) // otherLane is right of lane
                        {
                            if (dist < minDistRight) // closest lane right of lane is otherLane
                            {
                                minDistRight = dist;
                                lane.rightLaneForward = otherLane;
                            }
                        }

                        if (!lanesForward.Contains(lane) && !lanesReverse.Contains(lane))
                            lanesForward.Add(lane);
                        if (!lanesForward.Contains(otherLane) && !lanesReverse.Contains(otherLane))
                            lanesForward.Add(otherLane);
                    }

as:

if (isSameDirection) // same direction
                    {
                        if (cross > 0) // otherLane is left of lane
                        {
                            if (dist < minDistLeft) // closest lane left of lane is otherLane
                            {
                                minDistLeft = dist;
                                lane.leftLaneForward = otherLane;
                            }
                        }
                        else if (cross < 0) // otherLane is right of lane
                        {
                            if (dist < minDistRight) // closest lane right of lane is otherLane
                            {
                                minDistRight = dist;
                                lane.rightLaneForward = otherLane;
                            }
                        }

                        if (!lanesForward.Contains(lane) && !lanesReverse.Contains(lane))
                            lanesForward.Add(lane);
                        if (!lanesForward.Contains(otherLane) && !lanesReverse.Contains(otherLane))
                            lanesForward.Add(otherLane);
                    }

SampsonM8, 2021-09-06:
[image]

EricBoiseLGSVL, 2021-09-08:
Glad you found a solution.  I'll ping our team to look at what changes you made.  We have a few fixes for 2021.3 and it would be great to get this also.

RLitynskyy, 2021-09-09:
@SampsonM8 I looked into your suggested changes and looks like they are just hiding some deeper problem.
In our coordinate space cross product is using left hand rule instead of classical right hand as one axis is flipped. I checked on couple of known maps and current logic seems to be fine.
So I am guessing maybe problem is in input data itself.
Here is Simulator's coordinate system conventions just in case: https://www.svlsimulator.com/docs/getting-started/conventions/
Further investigation would require complete bug report with instructions on how to reproduce as well as all inputs used.

SampsonM8, 2021-09-13:
@RLitynskyy ，I am very glad to receive your suggestions.
I didn't say that any logic error exists in now lgsvl system. And I konw what you mean. 
But I think if any user has classical right hand Input , why not lgsvl takes some preprocess like transform pointcloud map to different kind of axes  to get a better experience ？

EricBoiseLGSVL, 2021-09-13:
The issue is that Unity is in the left hand coordinate system and you run into many issues trying to force in engine to a different one.  We try to make as many conversions when dealing with the Application, API and sensors output to expected right hand but when users are using the Editor to create content, the left hand needs to be used (until unity releases a way to change the editor) Sorry about the confusion and we understand the issues with it, we bang our heads on when to convert all the time. :)

SampsonM8, 2021-09-14:
Okay. Unity limits your work. 
Since the issue is caused by wrong input， it should be closed. :)
I am looking forward to your progress in map annotation. Thank you, EricBoiseLGSVL and RLitynskyy. :)

closed 2021-09-14 by EricBoiseLGSVL