lgsvl/simulator #1733: Question about VerticalRayAngles

duongtran11, 2021-09-03:
In LidarSensor.cs script, there is one line I don't understand
line 227 `if (VerticalRayAngles.Count == 0)`
In which situation this `(VerticalRayAngles.Count == 0)` would happen?

EricBoiseLGSVL, 2021-09-03:
If you look to LidarSensor.Templates, you see that only `Lidar32-NonUniform` has VerticalRayAngles, all others don't so the comment states that `if (VerticalRayAngles.Count == 0)` then Max/MinAngle needs to be calculated
// Assuming center of view frustum is horizontal, find the vertical FOV (of view frustum) that can encompass the tilted Lidar FOV.
// "MaxAngle" is half of the vertical FOV of view frustum.
Adding @kamkolak-lge for verification.

duongtran11, 2021-09-04:
Thank you for the answer. 
May I ask what is a non-uniform Lidar and what is it used for?

EricBoiseLGSVL, 2021-09-08:
No problem.  I'll ping the AD team on this