lgsvl/simulator #1734: Question about Lidar Sensor

lorlor, 2021-09-03:
Hi guys,

I have a problem about simulation point cloud sending. Limited by the rosbridge_websocket bandwidth, I couldn't get image with high resolution like 1920x1080 when point cloud sending with 32 lines. How could I send simulation point cloud with higher scanning lines?

Any advice will be appreciated.

My environment:
OS: ubuntu 16.04
Simulator version : 2021.1.1
Autoware: 1.12

EricBoiseLGSVL, 2021-09-03:
Are you using a gigabit switch/Router?

lorlor, 2021-09-07:
Sorry, I didn't have a giga device. Can I using UDP data frame to send simulation point cloud?

EricBoiseLGSVL, 2021-09-08:
No sorry, our docs state you need a high bandwidth switch/router.  UDP is even slower.  You can try it but we recommend the minimum requirements for data collection.  Anything slower is just too difficult to trouble shoot and we can't support.

lorlor, 2021-09-09:
Thanks for your reply.

closed 2021-09-09 by lorlor