lgsvl/simulator #1735: 3D Ground Truth sensor linear velocity is always 0 on y and z axis

ljjhome, 2021-09-03:
Hi
I'm trying to use the 3D Ground Truth sensor to do the trajectory prediction of surrounding vehicles.

I need the linear velocity of the surrounding vehicles.

I directly print the velocity as shown below:
std::cout << "obs vel : "<<it->velocity.linear.x<<" , "
                              << it->velocity.linear.y<<" , "
                              << it->velocity.linear.z<<std::endl;

and I get 0 on both y and z-axis :

[image]

I tried to modify the ego vehicle heading but it didn't help.

I would like to know how to get the full linear velocity vector? Did I miss some settings ? 

system : ubuntu 18.04
software : 2021.2.1
Thank you

EricBoiseLGSVL, 2021-09-03:
If you don't want linear velocity, why don't you use the velocity?

ljjhome, 2021-09-04:
Thanks for the reply.

I'm using the **ROS interface** directly with this msg:  [lgsvl_msgs::Detection3D message](https://github.com/lgsvl/lgsvl_msgs/blob/master/msg/Detection3D.msg) , and the "Velocity" in it is just the (speed , 0 , 0), and I found no other "Velocity" signal in this sensor msg that can provide the "Velocity".

Could you please tell me how to get the "Velocity" vector? 
(1) How can I get it **Directly from the ROS interface?**  if I missed something please tell me. I think it will be the most efficient way.
(2) or Do I have to use the python API? 
(3) or Do I need to read through the source code and recompile the whole software to get the "Velocity" vector?

ljjhome, 2021-09-06:
Hi @EricBoiseLGSVL ,
I dig into some source code these days, and maybe find a problem with it. I'm not familiar with c# and the lgvsl architecture, so please tell me if I'm wrong.

The code below is from [lgsvl/PerceptionSensor3D](https://github.com/lgsvl/PerceptionSensor3D/blob/be42a34ea94a2e0043d5f07b6a9d16ad57cff01b/PerceptionSensor3D.cs#L173)
we can see that:
LinearVelocity is a vector with zeros on the y and z-axis
velocity is the 3D velocity vector (maybe). 
[image]

The code below is from [lgsvl/ROS](https://github.com/lgsvl/ROS/blob/631ba206367347756204509077bbb61057d124e1/RosConversions.cs#L188) 
We can see that the ROS Interface uses the d.LinearVelocity variable which is the vector (speed,0,0) from the above code.
[image]

If what I found above is correct, Could you please change or add the variable for the 3D Velocity vector?
If it can only be modified in my own environment, Could you please show me some doc on how to compile and integrate these things?

Thanks, 
Best regards,
Jiajun

ljjhome, 2021-09-13:
Thank you @EricBoiseLGSVL, I did as you suggested, it works well.

closed 2021-09-16 by EricBoiseLGSVL