lgsvl/simulator #1736: Apollo Ros-bridge Car Does not move in simulator even when bridge status is connected

jp12AI, 2021-09-04:
I'm trying to simulate apollo 3.0 with LGSVL to test navigation mode (which seems to only be supported up to v3). I built the lgsvl-3.0 fork, the rosbridge connection, and tried to enable the relevant modules, however, there is no car view in dreamview even when the bridge status is "connected to client". I enable the relevant modules and get into the correct mode. I also cloned the lincoln vehicle under calibration to use as a vehicle for "ROS Apollo bridge" and even though it says connected nothing appears in dreamview. Am I missing something what order should I arrange the modules, what map, what vehicle. NOTE: In the PHOTOS the old-branch of 3.0 is shown however, I have tried with the most recent version and receive the same result with the 2015 jaguar and lincoln mkz. I ultimately just want to test navigation mode and it is possible to run on newer versions however this specific mode is mostly supported with 3.0 (for now). Any help/input would be appreciated Screenshots are below. Do I need to enable sim control maybe? 
[image]
[image]
[image]

EricBoiseLGSVL, 2021-09-13:
I'll ping @hadiTab but we are deprecating apollo 3.0 support and I recommend you don't use it.  Why do you need navigation mode?

jp12AI, 2021-09-14:
@EricBoiseLGSVL I want to use navigation mode because lidar is impractical and hd maps don't make any sense from a real perspective. I figured out how to use apollo 5.0 navigation mode with a modified script so this is no longer needed anymore. Thank you

closed 2021-09-14 by jp12AI