lgsvl/simulator #1737: Indianapolis Motor Speedway North and East coordinates are inverted

KyleRAnderson, 2021-09-05:
This is similar to #1312 in that the coordinates are off, but this time it appears to be in a different way.

The starting position's coordinate appears to be correct. However, we've noticed that moving North and South changes the Easting coordinate, and moving East and West changes the Northing coordinate, so the axes appear to be inverted somehow. We're observing this via the GPS Odometry sensor's `/lgsvl/odom_ego1` ROS 2 topic, and we've turned off `IgnoreMapOrigin` which should mean that the coordinates are absolute global coordinates according to the [documentation](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#gps-odometry).

I've attached a screen recording of the car moving along the North/South axis on the long straight of the Indianapolis motor speedway, as well as the corresponding rosbag which shows that the vehicle's X coordinate (Easting) position is what changes and not its Northing coordinate. The simulator version is 2021.2.2.

[GPS coordinates inverted.zip](https://github.com/lgsvl/simulator/files/7111765/GPS.coordinates.inverted.zip)

It's possible that we're doing something wrong here, so any help would be appreciated.

eandr127, 2021-09-06:
It seems that the conversion from Unity coordinates to GPS coordinates was changed in 2021.2 (looks like in commit 901aeeaf4df39c4d4746b7f8e52f1b1191a713dc, where the x and z coordinates are swapped). If the map wasn't updated for this change, then I think that this is the cause of the issue. I tested the IMS map on 2021.1.1, and the GPS axes were correct.

EricBoiseLGSVL, 2021-09-15:
We have found the issue.  Looks like MapOrigin.rotation.y was 90 not -90.  This will be fixed in the next release.

closed 2021-09-15 by EricBoiseLGSVL

sujithvemi, 2022-01-09:
> We have found the issue. Looks like MapOrigin.rotation.y was 90 not -90. This will be fixed in the next release.

I am facing the same issue while using a map that's built using 2020.3.3f1. Is it possible that the same bug exists in that release as well? Should I ask the map builder to update to latest release or will this be fixed?

sujithvemi, 2022-01-10:
> > We have found the issue. Looks like MapOrigin.rotation.y was 90 not -90. This will be fixed in the next release.
> 
> @EricBoiseLGSVL I am facing the same issue while using a map that's built using 2020.3.3f1. Is it possible that the same bug exists in that release as well? Should I ask the map builder to update to latest release or will this be fixed?
> 
> Edit: I was facing same issue with AutonomouStuff and BorregasAve map as well. So, I was thinking that it might be an issue in the simulator itself and not the map building process, because it was not just our custom map that is showing this.

Hey, I have figured out that the issue is with the map building process of ours, there was a missing transform to align it properly. So, I guess the same issue is pertaining to the other maps like AutonomouStuff and BorregasAve maps as well.