lgsvl/simulator #1738: How can I use Cloud Cluster (result visualizer)

Linsha350, 2021-09-06:
Hello.
I use LGSVL 2021.1 with lgsvl/apollo5.0.

First, to see point cloud, image, etc...,  I use cyber_visualizer in apollo.
However, the apollo 5.0 show error

> INFO: (09-06 09:19:36.036) Elapsed time: 1398.595s, Critical Path: 119.63s
modules/perception/proto/perception_camera.proto: warning: Import modules/common/proto/error_code.proto but not used.
INFO: Reading 'startup' options from /apollo/tools/bazel.rc: --batch_cpu_scheduling --host_jvm_args=-XX:-UseParallelGC --output_user_root=/apollo/.cache/bazel
WARNING: ignoring LD_PRELOAD in environment.
INFO: (09-06 09:19:46.968) Found 3749 targets...
INFO: (09-06 09:19:47.726) Elapsed time: 3.641s, Critical Path: 0.01s
1============================
[ OK ] Build passed!
[INFO] Took 1427 seconds
============================
Linsha@in_5_0_dev_docker:/apollo$ cyber_visualizer
Could not initialize GLX
Aborted (core dumped)

In apollo 6.0, the cyber_visualizer works. But, because my research is testing camera, lidar perception module, I must use apollo 5.0.
Any solution is not found.

Anyway, I saw visualization at test result in LGSVL.
https://www.svlsimulator.com/docs/user-interface/web/test-results-visualization/
[image]

However, the visualization is available only for cloud simulation

https://youtu.be/Da-pj1mZRlI?t=891
[image]

The youtube shows 'request cloud cluster' at 14:51
But, the wise svlsimulator does not show any cluster.
So, I use 'link to cloud' using my local computer.

How can I use cloud cluster for cloud simulation?
Or Do you have any solution to fix cyber_visualizer?

Thank you, always!

EricBoiseLGSVL, 2021-09-13:
Not sure why Apollo 5.0 DV is crashing on you.  I would open an issue on their forums.
Cloud is coming soon.  We are currently in development on this feature.

Linsha350, 2021-09-23:
Thank you for your reply!
I don't understand crash of Apollo 5.0 visualizer.
At issue of apollo github, someone install graphic driver in docker. But it is weird...
I'm waiting :)

closed 2021-09-23 by Linsha350