lgsvl/simulator #1740: Simulator bridge name attribute error

elmuhn, 2021-09-07:
Hi, 
I am getting the following error shown in both the image and the log file when I simulate AWFLexus2016RXHybrid with Autoware.Auto sensor configuration in CubeTown. I did not have this problem before the newest release. Also,  at the first attempt (right after launching the simulator for the first time), I can successfully simulate, for instance, Lincoln2017MKZ with Keyboard Control sensor configuration in the same map. Does anyone know the reason for this error? Thanks in advance for any help.

Nurullah

[image]

```
[CONN] Cloud URL https://wise.svlsimulator.com/
[CONN] Connecting to WISE
[CONN] Connected to WISE.
loader status: Idle->Loading 

[LOADER] Update simulation status Idle -> Loading
[CONN] Updated simulation e3cb327c-13b6-4f41-96a2-d9ef3cfd604e with status:Starting msg:
[CONN] POST /api/v1/simulations/e3cb327c-13b6-4f41-96a2-d9ef3cfd604e/status
Initialized the Simulation Network data.

found manifest at 24ad2e98-6732-4555-99b8-a8a526218abd/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x4887900.so
found manifest at 9f658040-e590-410b-abdd-38797987c1a2/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x48e6c00.so
found manifest at 8f9e9e34-0220-40aa-b85b-a6f0c45d7ed1/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x1b1d1c0.so
found manifest at b318f5c9-1aac-41f5-b597-080a7713ad7d/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x49446e0.so
found manifest at 9f658040-e590-410b-abdd-38797987c1a2/manifest.json

found manifest at 3d765dbb-6bb2-4f77-80f0-88bec7eeaf86/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x4947500.so
found manifest at dc5c631c-2f58-4db1-9fe6-06bc4e801bf4/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x4914750.so
found manifest at 391217cd-3ab1-4544-90ac-ea91c5f8f08f/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x49227f0.so
found manifest at 5f931d55-b797-40d5-b807-7a5892e7c0bd/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x49287d0.so
found manifest at 159b15b4-2100-4f7e-bf73-02a41e6b29dc/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x491a580.so
found manifest at 5ae06173-0ec5-45dc-95c3-1de27cb2869d/manifest.json

Fallback handler could not load library /home/elmuhn/svl/sim_svl_v21_2_2/simulator_Data/Mono/data-0x49354c0.so
All Downloads Complete

loader status: Loading->Starting 

[LOADER] Update simulation status Loading -> Starting
Unloading 5 Unused Serialized files (Serialized files now loaded: 12)
UnloadTime: 10.895388 ms
Initialized the Simulation scene for the network manager

found manifest at 24ad2e98-6732-4555-99b8-a8a526218abd/manifest.json

Bridge ROS2 already registered, ignoring duplicate(?) plugin

Failed to start 'CubeTown Autoware' simulation

MissingMethodException: string Simulator.Bridge.BridgeNameAttribute.get_Type()
  at SensorsController.InstantiateSensors (Simulator.Web.SensorData[] sensors) [0x0021d] in <6409380dfd5a476c90416dabe70eb910>:0 
  at SensorsController.SetupSensors (Simulator.Web.SensorData[] sensors) [0x00016] in <6409380dfd5a476c90416dabe70eb910>:0 
  at AgentManager.SpawnAgent (Simulator.AgentConfig config) [0x001c7] in <6409380dfd5a476c90416dabe70eb910>:0 
  at AgentManager.CreateAgentsFromConfigs (Simulator.AgentConfig[] agentConfigs) [0x0010e] in <6409380dfd5a476c90416dabe70eb910>:0 
  at AgentManager.SpawnAgents (Simulator.AgentConfig[] agentConfigs) [0x00000] in <6409380dfd5a476c90416dabe70eb910>:0 
  at SimulatorManager.Init (System.Nullable`1[T] seed) [0x003b4] in <6409380dfd5a476c90416dabe70eb910>:0 
  at Simulator.Loader.SetupScene (Simulator.Web.SimulationData simulation) [0x0007c] in <6409380dfd5a476c90416dabe70eb910>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Loader:SetupScene(SimulationData)
Simulator.<>c__DisplayClass44_0:<StartAsync>b__1(Boolean, String, String)
Simulator.<>c__DisplayClass46_0:<LoadMap>b__0(AsyncOperation)
UnityEngine.AsyncOperation:InvokeCompletionEvent()

loader status: Starting->Error string Simulator.Bridge.BridgeNameAttribute.get_Type()

[LOADER] Update simulation status Starting -> Error
[CONN] Updated simulation e3cb327c-13b6-4f41-96a2-d9ef3cfd604e with status:Error msg:string Simulator.Bridge.BridgeNameAttribute.get_Type()
[CONN] POST /api/v1/simulations/e3cb327c-13b6-4f41-96a2-d9ef3cfd604e/status
loader status: Error->Stopping 

[LOADER] Update simulation status Error -> Stopping
[CONN] Updated simulation e3cb327c-13b6-4f41-96a2-d9ef3cfd604e with status:Stopping msg:
[CONN] POST /api/v1/simulations/e3cb327c-13b6-4f41-96a2-d9ef3cfd604e/status
loader status: Stopping->Idle 

[LOADER] Update simulation status Stopping -> Idle
[CONN] Updated simulation e3cb327c-13b6-4f41-96a2-d9ef3cfd604e with status:Idle msg:
[CONN] POST /api/v1/simulations/e3cb327c-13b6-4f41-96a2-d9ef3cfd604e/status
[ANMGR] Initializing with TestReportId:6408e8e8-bd7f-407f-8d30-3fec9e66461e
[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

Unloading 57 unused Assets to reduce memory usage. Loaded Objects now: 6300.
Total: 16.486618 ms (FindLiveObjects: 0.593740 ms CreateObjectMapping: 0.319934 ms MarkObjects: 14.933788 ms  DeleteObjects: 0.637733 ms)

Unloading 2 Unused Serialized files (Serialized files now loaded: 0)
not running

[CONN] Updated simulation e3cb327c-13b6-4f41-96a2-d9ef3cfd604e with status:Stopping msg:stop requested but was not running simulation
[CONN] POST /api/v1/simulations/e3cb327c-13b6-4f41-96a2-d9ef3cfd604e/status
UnloadTime: 81.889908 ms

Unloading 71 unused Assets to reduce memory usage. Loaded Objects now: 4622.
Total: 17.627902 ms (FindLiveObjects: 1.822679 ms CreateObjectMapping: 1.366338 ms MarkObjects: 13.732793 ms  DeleteObjects: 0.702616 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 4622.
Total: 18.164906 ms (FindLiveObjects: 0.462473 ms CreateObjectMapping: 0.199286 ms MarkObjects: 17.377379 ms  DeleteObjects: 0.120598 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 4622.
Total: 20.953156 ms (FindLiveObjects: 0.637874 ms CreateObjectMapping: 0.659625 ms MarkObjects: 19.571259 ms  DeleteObjects: 0.077897 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 4622.
Total: 33.713394 ms (FindLiveObjects: 1.502746 ms CreateObjectMapping: 0.581107 ms MarkObjects: 31.600177 ms  DeleteObjects: 0.025789 ms)

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent

[CONN] Updated simulation e3cb327c-13b6-4f41-96a2-d9ef3cfd604e with status:Idle msg:
[CONN] POST /api/v1/simulations/e3cb327c-13b6-4f41-96a2-d9ef3cfd604e/status
[CONN] HTTP 200 OK
Receiving response of OK System.Net.Http.StreamContent
```

EricBoiseLGSVL, 2021-09-10:
Sorry you are having this issue.  Please un-link and relink the cluster.  Hopefully this will update the assets in the WebUI so you get the latest to run with 2021.2.2 tag release.  These assets have the method that is missing.  Let me know if that resolves the issue.

elmuhn, 2021-09-13:
> Sorry you are having this issue. Please un-link and relink the cluster. Hopefully this will update the assets in the WebUI so you get the latest to run with 2021.2.2 tag release. These assets have the method that is missing. Let me know if that resolves the issue.

Thanks for the reply.  I tried un/re-linking the cluster several times but unfortunately, it did not work. My academic advisor figured out that Autoware Car Control is causing the error by trying each sensor one by one. There seems to be a bug in the source code but I have not found it yet.

EricBoiseLGSVL, 2021-09-13:
Which simulator version are you on? 2021.2.2?  Are you making changes to the source at all?

elmuhn, 2021-09-14:
> Which simulator version are you on? 2021.2.2? Are you making changes to the source at all?

I have started getting this error with 2021.2.2 release. I was able to successfully run the simulation in 2021.2.1 but now I am getting the same error with older versions as well. I am looking through LGSVL Control Sensor and ROS2 bridge source code to find the problem.

jonteca, 2021-09-14:
I am getting roughly the same error.

elmuhn, 2021-09-14:
We solved the problem by modifying LGSVLControlSensor.cs in [here](https://github.com/lgsvl/LGSVLControlSensor). Just replace the if statements in OnBridgeSetup function (lines 124 and 142) with `if (bridge.Plugin.GetBridgeNameAttribute().Name == "ROS")` and `else if (bridge.Plugin.GetBridgeNameAttribute().Name == "ROS2")` respectively. Before modification, you should clone the repository inside Assets/External/Sensors. With modified source code, you should compile the sensor with the Unity Editor and upload it to the web ui as a new plugin. Also, do not forget to compile the simulator and restart the lgsvl_bridge before running the new simulation.

closed 2021-09-15 by elmuhn

EricBoiseLGSVL, 2021-09-16:
It seems this was a pull request that was merged without enough testing.  We are looking into this, thanks for posting.

jonteca, 2021-09-16:
Generally i am not sure if anyone has tested the ability to debug and use on Developer Settings.  Everytime i try i get an error with the Sensors not being found.

EricBoiseLGSVL, 2021-09-16:
First off @jonteca and @elmuhn if you need this change, you are not using 2021.2 branch but master and this is why you have an issue.  We are behind on master because we focused on the 2021.2.2 release. Please be sure you are using the 2021.2.2 release branch for simulator as well as all other cloned repos, not master.

@jonteca Developer settings is not for sensor debugging.  We use Developer Debug Mode for that.  This loads the local changes to a sensor so you don't have to upload to WebUI while testing in editor.  We still expect users to start from loader scene in Editor. Developer settings is mostly for artists to be able to run a sim quickly in editor.  It has improvements in the next release but again, it is not the designed work flow for sensor development.

jonteca, 2021-09-17:
This is on the 2021.2.2 release, will try and do you a repro.  Ah, so to clarify is there some debugging for sensors, is this documented anywhere?  This is what i am trying to do.

EricBoiseLGSVL, 2021-09-17:
Yes, please check.  Usually master is updated more often but we thought it was better to make 2021.2.2 better. 2021.3 is releasing soon that has many fixes on web and simulator for custom bridges.

Docs are [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/#testing-custom-assets-with-the-unity-editor)