lgsvl/simulator #1741: Signal sensor cannot detect traffic light in a map I build.

YoshikiKccs, 2021-09-08:
I learned how to do  map annotation. Therefore I combined  my map with  Cube Town.
I wanted to create horizontal signal, so I replace vertical traffic light with horizontal traffic light. Next I built that map and used it in simulator. But signal sensor cannot detect that traffic light. Please teach me what I should do.

[image]

heeen, 2021-09-13:
Have you linked the signal and lane/stop line? 
Looking at the source for the sensor, it seems to only detects signals that are child objects of the intersection that creates  the trigger collider volume
https://github.com/lgsvl/SignalSensor/blob/master/SignalSensor.cs

EricBoiseLGSVL, 2021-09-13:
Like @heeen stated, it is most likely an annotation issue.  Please post a screen shot of the intersection in Unity so we can see how you annotated it.

YoshikiKccs, 2021-09-14:
@heeen 
Thank you for teaching for me.
I will check that.

YoshikiKccs, 2021-09-14:
@EricBoiseLGSVL 
Thank you for working on solving the problem.
I see. I got it.
It is as shown below.
[image]
[image]

EricBoiseLGSVL, 2021-09-14:
Be sure the hdmap exporter supports the horizontal as well if you need that.

YoshikiKccs, 2021-09-14:
@EricBoiseLGSVL 
Thank you for your prompt response. I'm very sorry, but I don't know how to see the exporter.  I only know how to see the build settings window...  Could you please teach me how to see that hdmap exporter?

EricBoiseLGSVL, 2021-09-16:
what hdmap type are you planning to use?

YoshikiKccs, 2021-09-16:
@EricBoiseLGSVL 
I want to get hdmaps for Autoware Vector Map and Apollo 5 HD Map.

EricBoiseLGSVL, 2021-09-16:
No worries, if you look at the project panel on the bottom you can see a search panel.  Just type in AutowareMapTool or ApolloMapTool.  Open those scripts and look for signal in the code.  To add the new type, change the logic to test for the rotated signal and apply the data type the hdmap type expects, e.g., Apollo only had vertical supported when we wrote the exporter, look to apollo docs to see what they expect to have for the horizontal type and write a check (you could even add a string to the mapsignal class and check that name == horizontal). Then compile and export with your new functionality.

YoshikiKccs, 2021-09-17:
@EricBoiseLGSVL 
Thank you for telling me. I'll check them and add horizontal.

YoshikiKccs, 2021-09-22:
I'm sorry for asking again.
I've been looking for problem area and checked mapsignal class.
However, I couldn't find out.
If you like, please teach me more detail such as image...
[image]

EricBoiseLGSVL, 2021-09-22:
No worries. Looks like we already added different signal types, you just need to assign in the editor.  Be sure this code is similar or you may need to wait until next release.

[image]

[image]

[image]

[image]

YoshikiKccs, 2021-09-24:
@EricBoiseLGSVL 
Thank you very much for teaching me details. I'll try it!

YoshikiKccs, 2021-09-24:
@EricBoiseLGSVL 
I became to be able to detect horizontal signal. Thank you very much for your helping!!!
I really appreciate you!

EricBoiseLGSVL, 2021-09-24:
Glad to help

closed 2021-09-24 by EricBoiseLGSVL