lgsvl/simulator #1742: NMEA Sentences are malformed when the locale is not U.S.

MaxandreOgeret, 2021-09-08:
Hello, 
This problem appeared with the following configuration : 
- Sim running on **Windows** with **SVL v2021.2.2**
- Other computer was running the **ROS1 bridge**.

The windows computer was configured with Estonian locale with comma as decimal separator and it generated GPGGA sentence that was not in accordance with the [format](https://docs.novatel.com/OEM7/Content/Logs/GPGGA.htm).

I dont have the wrong sentence here but here's how it looked like from memory : 

```
$GPGGA,134658.00,5106,N,11402,W,2,09,1,1048,M,-16,M,08,
```

When switching the windows computer to U.S. locale and number format the problem was not here anymore. When reverting to Estonian format it went back.

Thanks for your time.

heeen, 2021-09-13:
Thank you, I believe I found the fix for this issue and I'm hoping we can include it in the next release.

EricBoiseLGSVL, 2021-09-16:
We have a fix for next release, thanks for posting and letting us know.

closed 2021-09-16 by EricBoiseLGSVL