lgsvl/simulator #1745: [BUG] Visual Studio won't/can't save files

domajstorovic, 2021-09-09:
I am unable to save any change to the existing file nor a new one. The first time I tried, I could actually save the scenario. However, ever since after the save button is just not responding. I have tried to reset the system, change the SVL version to 2021.1/2021.2.1., checked the read/write permission, and rerun the executable with admin rights. None of these worked.

System: Win 10 Enterprise, SVL 2021.2.2

domajstorovic, 2021-09-09:
Ok, so I have just tested the latest stable version (v2021.2.2) on Ubuntu and the Visual Editor is working without issues. I hope this will help isolate the cause on Win version.

EricBoiseLGSVL, 2021-09-13:
We haven't seen this issue before, we have many fixes and improvements next release

domajstorovic, 2021-09-14:
@EricBoiseLGSVL Ok, thanks! The Linux version is still working without issues, so I haven't experimented with Windows one, anymore. Have you set a release date maybe? Keep up the great work!

closed 2023-12-30 by domajstorovic