lgsvl/simulator #1746: Questions about add NPC agent of PythonAPI

joeking11829, 2021-09-09:
Hi LGSVL Teams,

I use PythonAPI to add NPC agent to simulator environment.
But I found some issues when I make NPC agent to follow waypoints.

1. The collision of NPC is not working, EGO vehicle could drive through NPC.
2. Some NPC stay on the position of spawn.
3. Some NPC could not finish all waypoints and stuck on road.

Any Idea to fix it ?
Thanks.

EricBoiseLGSVL, 2021-09-13:
1) yes, waypoint mode for moving NPCs uses triggers for collision callbacks, not physics colliders because they move differently
2) how are you spawning the NPC? Did you provide waypoints?
3) what do you mean stuck?  Did you make another api call?

Please provide more information and the api code you are using so we can debug further

joeking11829, 2021-09-22:
> 1. yes, waypoint mode for moving NPCs uses triggers for collision callbacks, not physics colliders because they move differently

Is there any way to enable physics colliders ?

> 2. how are you spawning the NPC? Did you provide waypoints?

I use following code snippets to create the NPC.
```
    def create_npc_vehicle(self, npc):
        # set agent info
        agent_type = lgsvl.AgentType.NPC
        agent_state = lgsvl.AgentState()
        agent_state.transform.position = lgsvl.Vector(npc.tx, npc.ty, npc.tz)
        agent_state.transform.rotation = lgsvl.Vector(npc.rx, npc.ry, npc.rz)

        # create waypoints
        npc_waypoints = self._create_npc_waypoints(npc.waypoints)
        
        # create agent
        npc_agent = self.sim.add_agent(npc.name, agent_type, agent_state)

        # check waypoints exist
        if npc_waypoints:
            npc_agent.follow(npc_waypoints, loop=True)

        return npc_agent

    def _create_npc_waypoints(self, npc_waypoints):
        lgsvl_npc_waypoints = []
        
        for data in npc_waypoints:
            effectors = []
            for effector in data['trigger']['effectors']:
                if 'typeName' in effector.keys():
                    effectors.append(lgsvl.TriggerEffector(effector['typeName'], effector['parameters']))
            trigger = lgsvl.WaypointTrigger(effectors)

            wp = lgsvl.DriveWaypoint(
                    position=lgsvl.Vector(data['position']['x'], data['position']['y'], data['position']['z']), 
                    speed=data['speed'], 
                    angle=lgsvl.Vector(data['angle']['x'], data['angle']['y'], data['angle']['z']), 
                    idle=data['waitTime'],
                    trigger=trigger,
                    )

            lgsvl_npc_waypoints.append(wp)

        return lgsvl_npc_waypoints

```
> 3. what do you mean stuck?  Did you make another api call?

After create the NPC, I just call "sim.run(time)" to start the simulation.
 
I create 10 ~ 20 NPCs in the simulation.
Is there any up bound for simulator ?

@EricBoiseLGSVL
Thank for the reply.

EricBoiseLGSVL, 2021-09-22:
1) you can change the simulator code but I don't recommend it.  Why are triggers collision callbacks not acceptable for collisions?
2) I don't see any waypoints created here.  See quick script 13-npc-follow-waypoints.py. Notice that the points are created here. It looks like your code is passing an empty set.
```
# This block creates the list of waypoints that the NPC will follow
# Each waypoint is an position vector paired with the speed that the NPC will drive to it
waypoints = []
x_max = 2
z_delta = 12

layer_mask = 0
layer_mask |= 1 << 0  # 0 is the layer for the road (default)

for i in range(20):
    speed = 24  # if i % 2 == 0 else 12
    px = 0
    pz = (i + 1) * z_delta
    # Waypoint angles are input as Euler angles (roll, pitch, yaw)
    angle = spawns[0].rotation
    # Raycast the points onto the ground because BorregasAve is not flat
    hit = sim.raycast(
        spawns[0].position + pz * forward, lgsvl.Vector(0, -1, 0), layer_mask
    )

    # NPC will wait for 1 second at each waypoint
    wp = lgsvl.DriveWaypoint(hit.point, speed, angle=angle, idle=1)
    waypoints.append(wp)
```
It also has examples of the collision callbacks.
3) We generally use 20-30 as max but it depends on sensors and map size/texture size.
Glad to help

joeking11829, 2021-09-24:
Thanks for the reply.

> 1. you can change the simulator code but I don't recommend it.  Why are triggers collision callbacks not acceptable for collisions?

I didn't set any trigger.
If I set trigger for NPC then collision will work ?

But I don't want to handle any collision callback, because these callbacks will interrupt the simulator and then the mechanism of record will failed.

> 2. I don't see any waypoints created here.  See quick script 13-npc-follow-waypoints.py. Notice that the points are created here. It looks like your code is passing an empty set.

I did create waypoints
```
def _create_npc_waypoints(self, npc_waypoints):
        lgsvl_npc_waypoints = []

        ......

        wp = lgsvl.DriveWaypoint(
                    position=lgsvl.Vector(data['position']['x'], data['position']['y'], 
                                  data['position']['z']), 
                    speed=data['speed'], 
                    angle=lgsvl.Vector(data['angle']['x'], data['angle']['y'], data['angle']['z']), 
                    idle=data['waitTime'],
                    trigger=trigger,
                    )

        lgsvl_npc_waypoints.append(wp)

    return lgsvl_npc_waypoints
```

Then make NPC follow waypoints.

```
# create waypoints
npc_waypoints = self._create_npc_waypoints(npc.waypoints)
        
# create agent
npc_agent = self.sim.add_agent(npc.name, agent_type, agent_state)

# check waypoints exist
if npc_waypoints:
    npc_agent.follow(npc_waypoints, loop=True)
```

EricBoiseLGSVL, 2021-09-24:
1) Python API waypoint follow does not have collision.  You will need to edit source to do this. We expect a trigger callback for collision to fail the simulation.
2) you have variables there but no way to see what the actual waypoints are.  You can see from my example that we calculate the waypoints from the spawn position.  I have no way to see the points you are using.  What is the `data` you are using? If you don't need trigger, why are you passing it?

joeking11829, 2021-09-24:
> 1. Python API waypoint follow does not have collision.  You will need to edit source to do this. We expect a trigger callback for collision to fail the simulation.

Got it, Thanks.

> 2. you have variables there but no way to see what the actual waypoints are.  You can see from my example that we calculate the waypoints from the spawn position.  I have no way to see the points you are using.  What is the `data` you are using? If you don't need trigger, why are you passing it?

Because I use VSE to configure NPC waypoints and triggers then save it to json file. 
I write a parser to convert the json file to ` data` .
So I could configure NPC waypoints and triggers via PythonAPI.

EricBoiseLGSVL, 2021-09-24:
Ah I see.  That's right you were using VSE to find waypoints.  I would double check the parser is working correctly and the data is good.  Then I would remove trigger params and see if that resolves some of the issues.  We have a few bugs in NPC waypoints that are solved in our next release, this might help also.

joeking11829, 2021-09-27:
> hen I would remove trigger params and see if that resolves some of the issues. We have a few bugs in NPC waypoints that are solved in our next release, this might help also.

Got it.
I'll try to remove trigger params.

Thank you.