lgsvl/simulator #1747: Radar and ROS1

risk-v, 2021-09-11:
Hi there,

I was wondering if it's possible to use the radar sensor with ROS1. I can see the radar 3D bounding boxes in LGSVL but when I tried "rostopic echo /radar" (/radar is my chosen topic using the Wise interface) the output it's empty. I am using LGSVL 2021.2. Thanks in advance.

[image]

MaxandreOgeret, 2021-09-12:
Last time I checked radar was not working properly. https://github.com/lgsvl/simulator/issues/1530

EricBoiseLGSVL, 2021-09-14:
radar is supported, search DetectedRadarObjectData in the code. In RosConversions.cs
```
public static ContiRadar ConvertFrom(DetectedRadarObjectData data)
        {
            var r = new ContiRadar()
            {
                header = new Header()
                {
                    sequence_num = data.Sequence,
                    timestamp_sec = data.Time,
                    module_name = "conti_radar",
                },
                object_list_status = new ObjectListStatus_60A
                {
                    nof_objects = data.Data.Length,
                    meas_counter = 22800,
                    interface_version = 0
                },
                contiobs = new List<ContiRadarObs>(),
            };

            foreach (var obj in data.Data)
            {
                r.contiobs.Add(new ContiRadarObs()
                {
                    header = r.header,
                    clusterortrack = false,
                    obstacle_id = obj.Id,
                    longitude_dist = UnityEngine.Vector3.Project(obj.RelativePosition, obj.SensorAim).magnitude,
                    lateral_dist = UnityEngine.Vector3.Project(obj.RelativePosition, obj.SensorRight).magnitude * (UnityEngine.Vector3.Dot(obj.RelativePosition, obj.SensorRight) > 0 ? -1 : 1),
                    longitude_vel = UnityEngine.Vector3.Project(obj.RelativeVelocity, obj.SensorAim).magnitude * (UnityEngine.Vector3.Dot(obj.RelativeVelocity, obj.SensorAim) > 0 ? -1 : 1),
                    lateral_vel = UnityEngine.Vector3.Project(obj.RelativeVelocity, obj.SensorRight).magnitude * (UnityEngine.Vector3.Dot(obj.RelativeVelocity, obj.SensorRight) > 0 ? -1 : 1),
                    rcs = 11.0,
                    dynprop = obj.State, // 0 = moving, 1 = stationary, 2 = oncoming, 3 = stationary candidate, 4 = unknown, 5 = crossing stationary, 6 = crossing moving, 7 = stopped TODO use 2-7
                    longitude_dist_rms = 0,
                    lateral_dist_rms = 0,
                    longitude_vel_rms = 0,
                    lateral_vel_rms = 0,
                    probexist = 1.0, //prob confidence
                    meas_state = obj.NewDetection ? 1 : 2, //1 new 2 exist
                    longitude_accel = 0,
                    lateral_accel = 0,
                    oritation_angle = obj.SensorAngle,
                    longitude_accel_rms = 0,
                    lateral_accel_rms = 0,
                    oritation_angle_rms = 0,
                    length = obj.ColliderSize.z,
                    width = obj.ColliderSize.x,
                    obstacle_class = obj.ColliderSize.z > 5 ? 2 : 1, // 0: point; 1: car; 2: truck; 3: pedestrian; 4: motorcycle; 5: bicycle; 6: wide; 7: unknown // TODO set by type not size
                });
            }

            return r;
        }
```
as for @MaxandreOgeret comment, we are looking into the issue posted.

PveOnly, 2021-11-07:
Hi is there some news ? I have the same problem.

EricBoiseLGSVL, 2021-11-08:
@PveOnly what issue do you have exactly?  Velocity is fixed and in the latest release 2021.3 and ROS1 is supported.
See RadarSensor.cs RosConversions.cs and Lgsvl.cs.

PveOnly, 2021-11-09:
Hi the issue is when I launch the rosbridge_websocket.launch to retrieve data from the radar I got these error pourring on the INFO tab : 
[image]

I'm using WSL Ubuntu 18.04 and the binary svlsimulator-windows64-2021.3 
I have no problem retrieving the LIDAR's data. When I try to retrieve the data from the radar I have no output in my terminal

EricBoiseLGSVL, 2021-11-10:
Please post the player log so we can see where the error starts.  I wonder if the WSL is an issue here but it could be the bridge.  Any idea @hadiTab ?

PveOnly, 2021-11-10:
No problem , here is the Player.log :
[Player.log](https://github.com/lgsvl/simulator/files/7513296/Player.log)
For this simulation I launched the simulation first then launched my rosbridge during the simulation (like 20 second after)

 For the sensor/bridge I used that : 
[image]

with "/radar" as a topics and "radar" as the frame for the radar and "/points_raw" and "velodyne" for the LIDAR on the default Jaguar2015XE

For the bridge with WLS I followed your tutorial (https://github.com/lgsvl/simulator/issues/754). I had some issue but I was able to resolve them

For the lgsvl_msg required for the radar I built from source because I could not install it with sudo apt: 
[image]

PveOnly, 2021-11-10:
The rosbridge terminal output when I launch it : 
[ros_bridge_terminal_output.txt](https://github.com/lgsvl/simulator/files/7513411/ros_bridge_terminal_output.txt)

EricBoiseLGSVL, 2021-11-10:
Can you clarify what you mean by built from source the lgsvl_msg? What was the issue?
Are you able to run the provided ROS bridge?
@revati-naik Are you able to reproduce this?

PveOnly, 2021-11-11:
The issue is that this line didn't work : 
[image]

[image]

So I just followed this instead : 
[image]

What is the provided ROS bridge ? The ROS bridge seems to work since I can retrieve data from the LIDAR

MaxandreOgeret, 2021-11-11:
Try with a dash and not an underscore

PveOnly, 2021-11-11:
Thank you I was able to install it with apt !
I reinitiated my WSL Ubuntu 18.04 installation and reinstalled the bridge then launched a new simulation to check if the problem came from that but unfortunately it didn't solve the problem.

I installed the melodic ros bridge by following these command : 
[melodic_ros_bridge_install.txt](https://github.com/lgsvl/simulator/files/7522475/melodic_ros_bridge_install.txt)

EricBoiseLGSVL, 2021-11-11:
We are looking into why the radar is not publishing, hope to have a solution soon

Ankur02, 2022-10-18:
Hi, I am still facing the issue where the /radar_data just gives empty objects. Has there been any resolution to the problem?