lgsvl/simulator #1751: NPC cannot go up from a slight uphill road, rolls back down

krtt, 2021-09-12:
I am using Python API test scenarios with our own map. On our map there is a slight uphill road (a bridge from real life map). When I put an NPC vehicle in front of the ego to follow closest lane, it drives (in this case 5mps) but cannot go up the small hill and rolls back down. This is unexpected. Is this - something SVL sim should consider? Or what do I need to do to make the NPC cross this bridge?

Small vid:

https://user-images.githubusercontent.com/22684441/132998484-b5793730-d8bc-41d9-a545-0732203289ee.mp4

EricBoiseLGSVL, 2021-09-13:
Follow closest lane just sets the lane speed.  We will look into this.  Can you show us the annotation for this section please?

krtt, 2021-09-20:
What do you mean by showing the annotation? I want to help but sorry, I am not too experienced here yet.

EricBoiseLGSVL, 2021-09-20:
No worries.  I am asking to screen shot the scene in Editor to see the annotation (open the annotation tool and select the section you are showing above).  I need more information to try to solve this issue.  NPCs in lane following are moved along the waypoints with rigidbody.moveposition and don't care about elevation.  Do you see this issue when running random traffic?