lgsvl/simulator #1752: Error when importing / exporting HD Map

NobushigeHaraki, 2021-09-13:
I want to convert an apollo HD Map to an OpenDRIVE map.
When I imported the apollo HD Map into SVL, I got the following error.
error:

-  No nearest intersection found for this stop sign 1041! Cannot assign it under an existing intersection.
- KeyNotFoundException: The given key was not present in the dictionary.

So I tried to delete the object "MapLineStop_1041" and add a new one, but I couldn't create a way point and couldn't create a stop line.
[image]

Also, when I exported this Map as an OpenDRIVE Map, the following error was output multiple times.
error:

-  Multiple junctionId found for one road predecessor / successor!

How can we solve these problems?

EricBoiseLGSVL, 2021-09-14:
We have many fixes for open drive importer that will be releasing very soon.  I'll add @RLitynskyy to see if they can give a few fixes in the meantime.

NobushigeHaraki, 2021-09-14:
Thanks!
Also, can you help me with the failure to create waypoints when importing Apollo HD Map?

EricBoiseLGSVL, 2021-09-16:
We will have to test with the new release.  Keep in mind simulator is not meant to be an hdmap conversion tool. We add extra data to the imported hdmap for NPCs and make optimizations to run better in Unity.

In order to create annotation data, you need a collider to raycast the temp waypoints.  You can add a flat plane and add a collider to it.