lgsvl/simulator #1753: 2D ground truth detections sometimes missing

niosus, 2021-09-14:
Hey everyone. I'm running SVL 2020.06 from within Autoware.Auto ade and I'm having an issue getting reliable 2D ground truth detections within the SVL. See the video below:

https://user-images.githubusercontent.com/1893598/133244066-3220b9d8-a66d-49e4-943a-b44ee96bdf62.mp4

Basically, some cars get detected sometimes but it is very flaky. As this happens within SVL I don't think it is relevant if I'm using ROS or which bridge I'm using, right? The configuration for the camera that I have looks as follows:
```json
{
    "type": "Color Camera",
    "name": "Main Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 2000,
      "Topic": "/simulator/main_camera",
      "Frame": "camera"
    },
    "parent": "base_link",
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  },
  {
    "type": "2D Ground Truth",
    "name": "2D Ground Truth",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 10,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "DetectionRange": 100,
      "Topic": "/simulator/ground_truth/detections2D"
    },
    "parent": "base_link",
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
  }
```

Any ideas here?

EricBoiseLGSVL, 2021-09-14:
We have large improvements to ground truth 2D in the upcoming 2021.3 release.  I recommend using this.  06 is very old at this point.

niosus, 2021-09-14:
With all due respect this is not really an answer. It is not up to me to pick which version to use and there are other issues popping up during any significant update. At this point, I would really like to know if there is anything I can do to have the data I need to be published. If the upgrade is the only solution to the problem I'm facing, the so be it, but I would like to know if there is an alternative.

EricBoiseLGSVL, 2021-09-15:
You are right, it is not really an answer, I apologize for not answering your issue.

Because of the multitude of code changes in simulator and sensors from 06 release I thought it better to suggest using our upcoming release because I am so excited about it.  We changed GroundTruth 2D and 3D sensors to no longer use crude box colliders but pixel match the bounding boxes to create an extremely accurate bounds.  It's fast and efficient and I hope others like it as much as a do.

As for the answer to your issue.
GT2D creates a box collider to detect collisions.  You could try increasing the detection range like so.
[image]

You can also remove the raycast to the center of the detected object, this is a poor occlusion test that we have also improved.
[image]

I do wonder, why can you not pick which version to use? If it is because you are following the Autoware.Auto docs? We have an up to date tutorial in our docs for current simulator and we are working together to update the one on Autoware.Auto as well. Again, 06 is very old and we don't guarantee support.

niosus, 2021-09-15:
Thanks a lot @EricBoiseLGSVL that helps somewhat. I do understand that it is an old version and support is not guaranteed, so thanks for bearing with me. We have an [issue](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/977) for looking at the compatibility of the new version SVL to Autoware.Auto. I have not looked myself and for now just assume there are reasons why we did not switch yet.

niosus, 2021-09-15:
I've struggled a little bit more and found that the reason was probably due to self-occlusion of the objects by the ego-vehicle. Once the camera is repositioned to avoid this we get all objects detected as expected. I believe this issue can be considered completely closed. I also think that if you guys moved away from the center-occlusion check then it won't pop up with the new version again.

closed 2021-09-15 by niosus

EricBoiseLGSVL, 2021-09-15:
Yep, really looking forward to the new release with updates to ground truth sensors.  @kamkolak-lge improved these a great deal.  I hope you can update.  We will continue to update functionality with Autoware.Auto so everyone can enjoy new features as best we can.