lgsvl/simulator #1754: SanFrancisco map and simulation do not match in version 2021.2.1 and 2021.2.2

bettertha, 2021-09-15:
I tried to follow the tutorial to connect apollo5.0 using both versions 2021.2.1 and 2021.2.2 .
The initial scene is:
[image]
The scene after the car went straight for a section of the road is as follow.
[image]

EricBoiseLGSVL, 2021-09-16:
Yes, this is a bug and will be fixed in the upcoming release.  Thanks for posting

closed 2021-09-16 by EricBoiseLGSVL

daohu527, 2022-03-22:
@EricBoiseLGSVL I know LGSVL is out of maintenance now, but pls can you tell me the specific cause of this problem and I will try to fix it

EricBoiseLGSVL, 2022-03-22:
We think this is from an issue with an outdated HDmap that Apollo is using or ignore map origin needs to be used.