lgsvl/simulator #1755: Client read/write failed causes problems

1Jyfc, 2021-09-15:
Hello SVL!
I'm recently working on test of Apollo with SVL. I need to carry on many times of test automatically, so I use PythonAPI to restart SVL again and again like follow:

```
sim = lgsvl.Simulator(SIMULATOR_HOST, SIMULATOR_PORT)
for i in range( MAX_ITERATION):
    sim.load("BorregasAve", case.seed)
    state = lgsvl.AgentState()
    state.transform = case.get_start_point() # customized start point
    ego = sim.add_agent(EGO_TYPE, lgsvl.AgentType.EGO, state)
    ego.connect_bridge(BRIDGE_HOST, BRIDGE_PORT)
    sim.add_random_agents(lgsvl.AgentType.NPC)
    print("Loading Dreamview...")
    dv = lgsvl.dreamview.Connection(sim, ego, BRIDGE_HOST)
    dv.set_hd_map('Borregas Ave')
    dv.set_vehicle('Lincoln2017MKZ')
    modules = [
        'Localization',
        'Transform',
        'Routing',
        'Prediction',
        'Planning',
        'Control'
    ]
    destination = case.get_destination() # customized destination
    dv.setup_apollo(destination.position.x, destination.position.z, modules)
```

And after I connect the bridge and run my code, such error may happen at some iteration:
```
E0915 22:25:51.096417 17166 client.cc:99] [cyber_bridge]Client write failed, disconnectingsystem:9
```
, which will cause some problems: sometimes the ego can't move after this happened, and sometimes the log in `apollo/data/log` will output wrong messages. These problems are important for me, so I wonder if it can be solved. I guess maybe it's because I restart the `sim` too many times, but I have to do that. Is there any suggestions? Hope for reply, thanks a lot.

EricBoiseLGSVL, 2021-09-16:
We have some bug fixes in the next release for api.reset.  I am guessing you have NPCs in the simulations you are running?  If so, this was not clearing correctly and has been fixed.  If not, please provide more details because I have not seen this error before.  Please provide simulator version, OS, api script so we can debug the issue.

1Jyfc, 2021-09-16:
Yes I add random NPC vehicles after each `sim.load`. I'm using version 2021.1.1 now, and if you mean that I should wait for your next release that you haven't publish yet, could you tell me when you'll publish it? Thanks.

And I think maybe my description is too simple and confusing, so I'll add some details.
Simulator version: 2021.1.1
OS: Ubuntu 18.04
Apollo: 6.0, github commit `45a787ab94e962505b9272a3a97284f9b48ed0a4`

I use `./script/bootstrap_lgsvl.sh` and `./script/bridge.sh` to connect to SVL, and during connection, the problem may happen on a certain reload of SVL:
[image]

The part of SVL in my api script is like the code I gave before. Each time I set different start points and destinations. And after 60s of `sim.run(60)` I'll do my analysis (not related to SVL) and reload simulator (not reset, because I need to change the random seed of NPC generation).

EricBoiseLGSVL, 2021-09-16:
We are testing the release now and it should be out within the next few weeks.
Are you completely stopping all Apollo modules before the next run? 

Any idea @revati-naik or @lemketron ?

1Jyfc, 2021-09-17:
Oh I tried and it really work. I reload the simulator without stopping modules, which caused that problem. I tried to stop and rerun modules and it works. Thanks a lot!

hyt1204, 2025-04-21:
> 哦，我试过了，它真的有效。我在没有停止模块的情况下重新加载模拟器，这导致了这个问题。我尝试停止并重新运行模块，它有效。多谢！

hyt1204, 2025-04-21:
> 哦，我试过了，它真的有效。我在没有停止模块的情况下重新加载模拟器，这导致了这个问题。我尝试停止并重新运行模块，它有效。多谢！

Hello, could you please tell me how to restart the modules? Can you provide me with the relevant commands? I'm not familiar with Apollo