lgsvl/simulator #1756: MapAnnotation ： Lane Width Compute Method

SampsonM8, 2021-09-15:
In OpenDriveMapExporter.cs File Function **CreateLaneWidths** is the way to calculate lane width.
And the pipeline is that Split left and right boundary into different partitions (resolution = 1)，calculate corresponding partition magnitude. Is this methold useful to curve?

SampsonM8, 2021-09-15:
@EricBoiseLGSVL

SampsonM8, 2021-09-16:
I mean maybe the result calculated by this method is not correct.

EricBoiseLGSVL, 2021-09-16:
Sorry, I am confused.  What is the issue exactly?  Have you found an issue with the CreateLaneWidths logic? We have many fixes to OpenDrive support in our next release. Please explain what the issue is so we can include in the next release if possible.

SampsonM8, 2021-09-17:
OpenDRIVE File exported by Pointcloud map annotation has bigger lane width error in curve road than part in straight road.

SampsonM8, 2021-09-17:
In curve part, the reference line matches map. In lane -1, the right boundary has inaccurate width. Its width is obviously bigger than actual lane width. In lane -2, the error is bigger. 
The above result doesn't appear in straight road part.

SampsonM8, 2021-09-17:
In curve part, 3.7m actual Lane width was annotated to about 4.7m nearly 5m.
Certainly, i annotated that boundary positions  are close to pointcloud data.

SampsonM8, 2021-09-17:
I tried to find answer by myself. So i looked up codes in  OpenDriveMapExporter.cs. I thought may partition resolution may be big and i tried to modified the number. But the issue still exists.
@RLitynskyy, i am also very very very very happy to receive your reply as soon.

SampsonM8, 2021-09-17:
I made a test just now. I splitted a long curve TrafficLaneSection into several short curve TrafficLaneSections. And the lane widths becomes normal in these short trafficlanesections.
I am confused. Was the appearance caused by accumulated error ？

RLitynskyy, 2021-09-22:
@SampsonM8 We are currently working on improving OpenDrive 1.4 format support. Right now OpenDrive import under reconstruction, export will be fixed after that. We are aware of export problems related to shape of an output geometry, thanks for bringing this up.
If you have any thoughts and findings that you are ready to share, would be happy to integrate them if they work correctly.