lgsvl/simulator #1757: I could not simulator

roderauto, 2021-09-16:
Hi,
I couldn't stop the simulation.
Please help.
I'm running in a Unity development environment

[image]

I tried
・ Delete cache
・ Log out
・ Restart Unity
・ Restart your PC

EricBoiseLGSVL, 2021-09-16:
Please use Ctrl+F5 in the WebUI with Unity Hub and Unity Editor completely closed.  This issue is fixed in the next release.

roderauto, 2021-09-22:
I tried but couldn't stop.
Is it because you deleted the previous cluster?

EricBoiseLGSVL, 2021-09-23:
Are you stopping the sim in the application?  I recommend stopping in the application until next release.  There is a bug with stopping in webui.  What about deleting cluster?  Are you deleting the cluster while running the simulation?

roderauto, 2021-09-28:
I was able to stop the simulator. thank you！

closed 2021-09-28 by roderauto