lgsvl/simulator #1759: G29 doesn't control the car in Windows64 2020.05 version

chihchunhsu-dev, 2021-09-17:
Hello community,

I am new to LGSVL. Our team has been using LGSVL 2020.05. We bought a G29 steering wheel and tried to run it in our simulation. 
However, even though the Info panel says G29 is connected, when we step on the gas pedal or turn the wheel the car in the scene doesn't move at all. 

[image]

Does anyone ever get this problem? How did you solve it? Any advice would be appreciated.

Thanks,
Chih-Chun

EricBoiseLGSVL, 2021-09-17:
Sorry, 05 release is very old, we recommend you update to latest simulator.  We are about to release 2021.3 that will allow you to add your own Unity input controls to a sensor.
If you can't wait/update then you will need to look at the input controls.  That version of sim is using a beta input package and might need changes.  Open the unity editor and make sure the G29 is correctly installed.  You will need the manufacturer software and drivers.  Unity editor has an input debugger. make sure all inputs are received and  the controls are mapped.

These are the windows to edit.  Make the fixes and just build a new simulator binary
[image]
[image]
[image]

chihchunhsu-dev, 2021-09-23:
Thank you it worked!

closed 2021-09-23 by chihchunhsu-dev