lgsvl/simulator #1760: cuda container version

usydptc, 2021-09-17:
Here there,

I'm new to SVL Simulator and trying to follow the user guide "Running latest Apollo with SVL Simulator". When I launch the container, it says "**requirement error: unsatisfied condition: cuda>=11.1**". I have updated my cuda into 11.1 and trying to update the cuda container into 11.1. But from the installation steps for the NVIDIA Container Toolkit, I think it just gives the step for container 11.0 installation because the test command in the guide is like "**cuda:11.0**-base nvidia-smi".

So I'm wondering is there some suggestions about updating container 11.0 into higher version?
Cheers,
ptc

EricBoiseLGSVL, 2021-09-17:
Are you looking for [this](https://developer.nvidia.com/cuda-11.1.0-download-archive)?

usydptc, 2021-09-18:
> Are you looking for [this](https://developer.nvidia.com/cuda-11.1.0-download-archive)?

Hi, many thanks for your reply. I have downloaded cuda toolkit 11.1, but now my nividia-docker2 still saying the cuda version is 11.0. On the other hand, by using "nvidia-smi" command, it says cuda version is 11.1

So do I need to update the nividia-docker2 or is there some ways to let nividia-docker2 find cuda 11.1 but not 11.0?

EricBoiseLGSVL, 2021-09-20:
[This](https://github.com/NVIDIA/nvidia-docker/issues/1237) seems like a good explanation of the issue you are seeing.

closed 2021-09-22 by usydptc