lgsvl/simulator #1763: NPCs not adding in SVL 2021.2 version

MartinaB85, 2021-09-22:
Hi,
I'm currently using the 2021.2 version of the SVL simulator.
I'm trying to add several NPCs through python code but when playing the simulation in both API only (as suggested in another issue https://github.com/lgsvl/simulator/issues/1336) and Python API I get the error:

    npc2 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)
  File "/home/martina/Desktop/PythonAPI/lgsvl/utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "/home/martina/Desktop/PythonAPI/lgsvl/simulator.py", line 162, in add_agent
    uid = self.remote.command("simulator/add_agent", args)
  File "/home/martina/Desktop/PythonAPI/lgsvl/remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Object reference not set to an instance of an object

I notice that in this version of the simualtor the folder NPC under Assets/Prefab disappeared. Even adding the same folder from version 2021.1 and the related ones and replacing NPCSettings.asset and NPCManager.prefab it is still not working.

How can I add the same NPCs available in the other versions of the simulator to the 2021.2 ?

Thank you very much,

Martina

EricBoiseLGSVL, 2021-09-22:
NPCs have moved to a separate repo called DefaultNPC.  If you are using the Unity Editor, you need to clone (enable gitlfs!) and build so they are in the AssetBundles folder.  If you are using the binary we provide, the npcs are already there.  You can also copy the npcs from the binary and place in the cloned simulator directory AssetBundles/.  See docs [here](https://www.svlsimulator.com/docs/plugins/npc-plugins/) and repo [here](https://github.com/lgsvl/DefaultNPC).

MartinaB85, 2021-09-28:
Thank you very much for your help :)

closed 2021-09-28 by EricBoiseLGSVL